FreeCAD/FreeCAD #6323: [Sketcher] bugs in feature "Inserting knots in B-Spline"

donovaly, 2022-02-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

#5311

### Version

0.20 (Development)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.27422 (Git)
Build type: Release
Branch: master
Hash: d938733eaf2c2ce7cb18d1cbb56147185c473530
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

Sketcher

### Issue description

There are bugs in this feature (PR #5311):
- it is still not yet documented in our Wiki ;-), please do so
- the dialog always shows "0.5000000", so it ignores the system-wide decimal place settings
- the dialog's spin edit step is too large. It is set to 1.0. This is too large because using 1.5 or -0.5 leads for most cases to errors.
- the tooltip says that the feature will insert a knot with the given parameter and multiplicity, but it is not clear what the parameter is about. It is also not possible to specify the multiplicity.

### Anything else?

@ajink

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-02-08:
PR for fix at #5517

AjinkyaDahale, 2022-02-08:
@luzpaz #5517 does not fix the issues mentioned here

AjinkyaDahale, 2022-02-08:
@donovaly thanks for the comments. I already inserted a line in the changelog for 0.20 about this feature but will wait till more features are added (like clicking on the curve to pick a point if possible). For the rest of the issues you mentioned I'll try to fic them over the coming days, but you are welcome to suggest fixes if you have the time.

AjinkyaDahale, 2022-02-08:
@donovaly regarding point 2 of your issue:
1. Where can I find the decimal information?
2. Apart from the number of digits after the separator, are you also referring to the decimal separator itself? Because I'm not sure how I can change the latter.

AjinkyaDahale, 2022-02-08:
See #6326 

>     * it is still not yet documented in our Wiki ;-), please do so

There is a statement in [0.20 release notes](https://wiki.freecadweb.org/Release_notes_0.20#Further_Sketcher_improvements).
The wiki page itself is on the TODO list. Some features I'd like to add before writing it up:
1. There is a point temporarily drawn on the viewport showing the new knot. 
2. The parameter changes when we click on the b-spline to indicate where we want the knot.

If someone has a suggestion for how to get this done I'd be grateful.

>     * the dialog always shows "0.5000000", so it ignores the system-wide decimal place settings

Waiting for response to the comment above.

>     * the dialog's spin edit step is too large. It is set to 1.0. This is too large because using 1.5 or -0.5 leads for most cases to errors.

Changed it to 0.001.

>     * the tooltip says that the feature will insert a knot with the given parameter and multiplicity, but it is not clear what the parameter is about. It is also not possible to specify the multiplicity.

Removed the mention of multiplicity in that context. I did not add an option to specify multiplicity because
1. We can change that using the "Increase multiplity" command.
2. Adding more complex behavior would need to create a custom dialog box (as opposed to using `QInputDialog`). I might eventually need to do that as I add the preview feature mentioned above.

donovaly, 2022-02-08:
I am traveling, thus cannot compile.
Concerning the decimals, the mistake is that you hardcode to:
`knotDialog.setDoubleDecimals(8);`

Search in The Part/Gui code for "XDirectionEdit" so see how it is done there.

AjinkyaDahale, 2022-02-09:
There's a stub now in the wiki for this feature: https://wiki.freecadweb.org/Sketcher_BSplineInsertKnot. The navigation still needs to be resolved.

donovaly, 2022-02-12:
> There's a stub now in the wiki for this feature: https://wiki.freecadweb.org/Sketcher_BSplineInsertKnot. The navigation still needs to be resolved.

I already fixed the navigation few days back. What I miss in the Wiki page its to describe how it exactly works. For example I have a simple spline with only the end knots and want to insert a knot. What does then e.g. "0.5" as parameter mean? You can referenced in the Wiki already this page:
https://wiki.freecadweb.org/B-Splines
So you can use this reference also for your description/explanation.

donovaly, 2022-02-12:
I trust you that you will extend the Wiki page ;-) and close the issue as resolved.

donovaly, 2022-02-12:
.

closed 2022-02-12 by donovaly

AjinkyaDahale, 2022-02-12:
> I trust you that you will extend the Wiki page ;-) and close the issue as resolved.

Targeting after #6348 is completed and merged.