FreeCAD/FreeCAD #6324: [Bug] [Sketcher] Segmentation fault on knot insertion in B-Spline

AjinkyaDahale, 2022-02-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=568723#p568723

### Version

0.20 (Development)

### Full version info

```shell
OS: Manjaro Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.27445 (Git)
Build type: Release
Branch: master
Hash: 4f3e4eff03e156ed80f310b9fe16c6c53fb54b55
Python version: 3.10.2
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Sketcher

### Issue description

See forum post for explanation and examples. To summarize, if we already have a B-Spline `spline1` in a sketch and we create another spline `spline2`, then using the "Insert Knot" feature on `spline1` results in a segmentation fault.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AjinkyaDahale, 2022-02-08:
Fix suggested in PR #5517

AjinkyaDahale, 2022-02-09:
Segmentation fault also appears to happen when adding a second knot even if there is only one B-Spline. I don't have a build at hand to confirm whether or not #5517 fixes that.

wwmayer, 2022-04-17:
Today I got a crash when following the instructions as described in the forum. Apparently the issue is not related to the knot insertion but the creation of the new B-Spline.
With the address sanitizer option it told me there was a heap-buffer overflow in the function `DrawSketchHandlerBSpline::drawCursorToPosition`

Checking why this happens revealed that the vector BSplinePoles is empty. Thus the call of
```
float length = (position - BSplinePoles.back()).Length();
```
leads to undefined behaviour. So, from this point on everything can happen. The program can crash immediately, it can crash all the sudden at a later point or it can continue.

See 139370a20f

AjinkyaDahale, 2022-04-17:
@wwmayer did you reopen this or was this never closed? @donovaly maybe this should also be on milestone if it still exists.

wwmayer, 2022-04-17:
> did you reopen this or was this never closed?

It was never closed. I was looking through some open issues and found this one.

donovaly, 2022-04-18:
> maybe this should also be on milestone if it still exists.

If this commit: https://github.com/FreeCAD/FreeCAD/commit/139370a20fd
does not fix the segfault then yes.

AjinkyaDahale, 2022-04-18:
There is one last remaining thing:
1. Begin creating a b-spline and add a few points.
2. Hover over the last circle or center and press backspace to delete it.
3. Press LMB without moving cursor. 

There will be redundant constraints. They can be selected and deleted, but they exist.

donovaly, 2022-04-18:
I cannot test FC 0.20 here (for the next week). If there is a regression to 0.19 or a crash, please add the milestone 0.20.

AjinkyaDahale, 2022-04-18:
> I cannot test FC 0.20 here (for the next week). If there is a regression to 0.19 or a crash, please add the milestone 0.20.

Since it relates to a new feature (deleting last pole) it can't be a regression.

AjinkyaDahale, 2022-04-21:
> > There will be redundant constraints. They can be selected and deleted, but they exist.
> 
> Is this already reported?

I don't think so. No mention in the forum on a cursory look.

AjinkyaDahale, 2022-06-08:
> bump on this ticket

The original issue appears to be resolved so this can be closed. The redundant constraints issue still exists.

closed 2022-06-08 by chennes