FreeCAD/FreeCAD #6329: Draft Grid size  not changin to draft preferences menu [Bug] <title>

sedatkrn, 2022-02-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=26277

### Version

0.20 (Development)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.27428 (Git)
Build type: Release
Branch: master
Hash: 27460358508a2057e0ec57a418641435f12628dd
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Turkish/Turkey (tr_TR)
```

### Subproject(s) affected?

Draft

### Issue description

When I open the features menu to change the grid dimensions. Despite the value I entered all kinds of unreasonable grid dimensions.

[image]

### Anything else?

>>> ### Begin command Std_Workbench
>>> Gui.activateWorkbench("DraftWorkbench")
>>> ### End command Std_Workbench
>>> Gui.runCommand('Std_DlgPreferences',0)
>>> Gui.runCommand("Draft_ToggleGrid")
>>> Gui.runCommand("Draft_ToggleGrid")
>>> Gui.runCommand('Std_DlgPreferences',0)
>>> Gui.runCommand('Std_DlgPreferences',0)
>>> Gui.runCommand('Std_About',0)
>>> Gui.runCommand('Std_DlgPreferences',0)
>>> Gui.runCommand('Std_DlgPreferences',0)
>>> Gui.runCommand("Draft_ToggleGrid")
>>> Gui.runCommand("Draft_ToggleGrid")
>>> Gui.runCommand('Std_DlgPreferences',0)
>>> Gui.runCommand("Draft_ToggleGrid")
>>> Gui.runCommand("Draft_ToggleGrid")
>>> 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2022-02-11:
At the moment the Draft workbench doesn't see when the preferences are changed. See https://forum.freecadweb.org/viewtopic.php?f=8&t=26277&p=567016&hilit=draft+working+plane+preferences+observer#p567016

sedatkrn, 2022-04-16:
[code]
OS: Ubuntu 21.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.
Build type: Release
Branch: unknown
Hash: 93d745235cdccce8652e6eb97309f7b2764bf367
Python 3.9.7, Qt 5.15.2, Coin 4.0.0, OCC 7.5.2
Locale: Turkish/Turkey (tr_TR)
Installed mods: 
  * sheetmetal 0.2.49
  * BOLTSFC
  * BIM 2021.12.0
  * fasteners 0.3.40
  * A2plus 0.4.56
  * dodo
  * fcgear 1.0.0
[/code]

The bug is still going on..

[image]

luzpaz, 2022-04-16:
bumped the [thread](https://forum.freecadweb.org/viewtopic.php?p=589093#p589093)

sedatkrn, 2022-04-22:
[code]
OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.28730 (Git)
Build type: Release
Python 3.8.13, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: Turkish/Turkey (tr_TR)
Installed mods: 
  * A2plus 0.4.56
  * BIM 2021.12.0
  * CurvedShapes 1.0.3
  * Curves 0.4.2
  * dodo
  * fasteners 0.3.40
  * fcgear 1.0.0
  * sheetmetal 0.2.49
[/code]

[image]
[image]

 Bug is still going on :(

yorikvanhavre, 2022-04-22:
This requires an observer to be added (ref https://forum.freecadweb.org/viewtopic.php?p=564680#p564680 ) but I am not sure it is wise to be done so close to the release as it can introduce more bugs. In the meantime, we could either:
1) Add a text to the appropriate Draft preferences page telling users to turn the grid off/on to apply changes
2) Remove grid settings entirely from the preferences page (let users manage them only via the working plane button)

jpmeijers, 2022-05-30:
I'm also affected. Turning grid off/on does not help. Even restarting FreeCAD doesn't apply the new settings.

[image]

```
OS: Ubuntu 20.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.28945 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0fb7cd9)
Hash: 0fb7cd9758d850f536d571cc094c9833fee18fff
Python 3.9.13, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: English/South Africa (en_ZA)
```

sedatkrn, 2022-06-09:
Problem solved  :))))))))))))))))

[code]
OS: Linux Mint 19.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.
Build type: Release
Branch: unknown
Hash: b9b21ddac562f03cef162bb1f7feb4eea3de8b1e
Python 3.6.9, Qt 5.9.5, Coin 4.0.0a, OCC 7.5.2
Locale: Turkish/Turkey (tr_TR)
Installed mods: 
  * fcgear 1.0.0
  * dodo
  * Behave-Dark-Colors 0.0.1
  * Assembly3 0.11.3
  * Assembly4 0.11.12
[/code]

closed 2022-06-09 by sedatkrn