FreeCAD/FreeCAD #6330: Number formatting customization

0penBrain, 2022-02-09:
The last info I have is that this issue is not resolved:
- take me as a German, my Windows is a German one, therefore it uses the comma as decimal separator.
- I use FreeCAD in English and therefore want the dot as decimal separator

As it is I have no setting to achieve this.

_Originally posted by @donovaly in https://github.com/FreeCAD/FreeCAD/issues/5871#issuecomment-1031895248_

donovaly, 2022-02-09:
As in other program, I expect that when I change the language to English, I get
- English menus ->works
- I enter e.g. "5.2 m" I get 5.2 m, but I getting 5200 m -> works not

I am currently working with another person on a sketch and since he cannot speak German, he sets the language to English. It is annoying for him, that he nevertheless mus use the comma to input the decimal sign. (We only have German Windows machines here.)

> FreeCAD to apply same regional number format as selected language in preferences ?

Yes. This is also the intuition.

> An option to enforce usage of C local ?

What do you mean with "C local"? I never heard about this term yet.

0penBrain, 2022-02-09:
> * I enter e.g. "5.2 m" I get 5.2 m, but I getting 5200 m -> works not

What is the relationship with language or number formatting ? Unit scaling has nothing to do with that. ;)

> > FreeCAD to apply same regional number format as selected language in preferences ?
> 
> Yes. This is also the intuition.

OK, this may be a bit picky hence my below proposal. Will look what can be done.
> > An option to enforce usage of C local ?
> 
> What do you mean with "C local"? I never heard about this term yet.

Roughly "C" (aka "POSIX") locale is a regional setting close to English, but with some differences like simplified number formatting.
I proposed this simplified setting because I think this is main encountered issue. People wanting to enforce usage of "English like" number formatting. C locale has the advantage to be standard across platforms.

0penBrain, 2022-02-10:
@donovaly I pushed PR #6335 to allow enforcing C locale usage in FreeCAD. You can eventually test it.
In between it's merged, most probably you can solve the problem of your colleague by typing into Python console
```
from Qt import QtCore
QtCore.QLocale.setDefault(QtCore.QLocale.c())
```
Will work on setting number format according selected UI language later on.
HTH

adrianinsaval, 2022-02-10:
> > * I enter e.g. "5.2 m" I get 5.2 m, but I getting 5200 m -> works not
> 
> What is the relationship with language or number formatting ? Unit scaling has nothing to do with that. ;)

when you input the incorrect decimal separator freecad gives a bigger number instead of the desired one, maybe he slipped some extra 00 for dramatic effect :)
For a similar real example, if you input 5.125m you would instead get 5125m

I agree with uwe that the expected behavior would be to have the same as the language chosen
It would also be nice to be able to specify the decimal and thousands separators manually as in for example excel

on a separate note, I found an issue with decimal point replacement, it also replaces the point with a comma in the python console and macro editor, but of those are not locale dependent on the number interpretation, it should always be a point. Should I create a separate issue for this?

0penBrain, 2022-02-10:
> when you input the incorrect decimal separator freecad gives a bigger number instead of the desired one, maybe he slipped some extra 00 for dramatic effect :) For a similar real example, if you [...]

I explained this a dozen times and I'm tired to do it.
BTW @donovaly are you sure ? From what we know, '5.2m' should give '52m' and not '5200m'.

> I agree with uwe that the expected behavior would be to have the same as the language chosen It would also be nice to be able to specify the decimal and thousands separators manually as in for example excel

Yep, but that's much work. As I said, I'll eventually work on this later on. In between, enforcing C locale is a good solution for most cases IMO (in combination with separator substitution for German)

> on a separate note, I found an issue with decimal point replacement, it also replaces the point with a comma in the python console and macro editor, but of those are not locale dependent on the number interpretation, it should always be a point. Should I create a separate issue for this?

Honestly, I now this limitation for a few time and have thought about it. I still have questions in mind to know if it should an option or not, and what is the best way to implement. You can open an issue if you want.

donovaly, 2022-02-13:
> What is the relationship with language or number formatting ? Unit scaling has nothing to do with that. ;)

Let's take for example LibreOffice:
[image]

The language defines the rules. So German uses comma as decimal separator, English dots. etc. So all we need is an option when one uses FC in English, that one can also takes the decimal sign the language/locale defines.

commit e3bc06a3 (2022-03-17): Gui: add preference to use selected language number formatting, fixes #6330

 If not enabled (default), will defaults to C/POSIX formatting
closed 2022-03-17 by donovaly