FreeCAD/FreeCAD #6352: [Bug][Mac] Cmd+Z in Spreadsheet is interpreted in ComboView->Model as well

elosha, 2022-02-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

I'm not using the forums

### Version

0.19 (Release)

### Full version info

```shell
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.19.24366 (Git)
Build type: Release
```

### Subproject(s) affected?

Spreadsheet

### Issue description

In several cases it happened, if I handled spread sheets via the ComboView->Model tree (creating and deleting items), then edited a spreadsheet and pressed Cmd+Z (Undo), the command was also evaluated by the Model tree, thus undoing creation/deletion of items like spreadsheets. It once even happened that I deleted the spreadsheet I was currently working on because I undid its creation.

I guess this might be a GUI focus problem.

Furthermore but unrelated, it happens that an Undo in the spreadsheet undoes several lines at once and left me with the Redo function disabled.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-02-13:
@elosha we ask users (especially those that are using the stable (v0.19) and especially on macOS) to post their issues first to the forum because there are users who use the FreeCAD development version (thousands of commits ahead of stable) that can test to see if the bug still persists. It saves the bug triagers and developers time. Are you willing to do this?

donovaly, 2022-02-14:
Please first open a forum thread (you can always use the general Help forum).

When people there agree there is a bug, the issue can be reopened.
Issues without prior forum discussion won't be accepted since otherwise out issue tracker would quickly overflow and unusable.