FreeCAD/FreeCAD #6354: [Bug] FEM - GMSH 'hwall_t' boundary condition finally removed in GMSH 4.6.0

StefanBruens, 2022-02-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?p=508883&sid=6d38c8a58aed8b663a5c2cd73ad115d0#p508883

### Version

0.20 (Development)

### Full version info

```shell
OS: openSUSE Tumbleweed (KDE//usr/share/xsessions/plasma5)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.Unknown
Build type: Release
Python version: 3.8.12
Qt version: 5.15.2
Coin version: 3.1.3
OCC version: 7.5.3
Locale: English/United States (en_US)
```

### Issue description

Open Cantilever example, remesh with GMSH, add BoundaryLayer on fixed (constrained) face. Remesh ...

```
Gmsh had warnings ...
Error   : '/tmp/fcfem_ywt6vnqc/shape2mesh.geo', line 17: Unknown option 'hwall_t' in field 1 of type 'BoundaryLayer'
```

https://github.com/FreeCAD/FreeCAD/blob/2a79ce4bdc3113a03d3c9bb2d30968ace4766fc0/src/Mod/Fem/femmesh/gmshtools.py#L622

See https://gitlab.onelab.info/gmsh/gmsh/-/issues/913

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

StefanBruens, 2022-02-14:
As far as I can see, BoundaryLayers in GMSH are only supported for 2D Meshes (or 3D meshes extruded from these, directly in GMSH).

See https://gitlab.onelab.info/gmsh/gmsh/-/issues/811

linuxguy123, 2023-03-28:
I encountered this error.   It would be helpful if the user was notified that boundary meshes can only be used on 2D meshes.

PR #15566 (2024-07-22): Fem: Remove unused gmsh option - fixes #6354
Fixes #6354

@FEA-eng
commit 00b8e11d (2024-07-22): Fem: Remove unused gmsh option - fixes #6354
closed 2024-07-22 by chennes