FreeCAD/FreeCAD #6358: [Bug] TreeView selection of Arch Window not possible

jipem01, 2022-02-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

/

### Version

0.19 (Release)

### Full version info

```shell
OS: Windows 10 Version 2004
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +99 (Git)
Build type: Release
Branch: Branch_0.19.3
Hash: 6530e364184ce05ccff39501e175cf2237e6ee4b
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: French/France (fr_FR)
```

### Subproject(s) affected?

Arch

### Issue description

Errors when trying to select [Arch] windows in treeview.

In Report windows i got message :
23:39:41  922.312 Selection.cpp(1666): Sub-object Localjo#Part.Wall.Windows. not found

In python console, 
Gui.Selection.addSelection('Localjo','Part','Wall.Windows.') give same error message
But
Gui.Selection.addSelection('Localjo','Part','Wall.Windows') is okay.

So either is something wrong with selection.cpp for windows, or with the treeview.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-02-13:
Most likely fixed in v0.20dev  
please either ask on the forum or install v0.20dev and test

jipem01, 2022-02-13:
Additionnaly, if i move the wall out of a part, the error disapear.
When trying windows to select in treeview,
Gui.Selection.addSelection('Localjo','Part','Wall.Windows')
become Gui.Selection.addSelection('Localjo','Wall.Windows') and works well!

yorikvanhavre, 2022-02-14:
I tried this:

* create a wall
* create a window on one face of the wall
* create a part
* move the wall (and the window) to the part

and I can select the window in the tree view without any visible error, the window is selected, not the part.

Can you provide exact step-by-step instructions to reproduce your problem?