FreeCAD/FreeCAD #6366: [Feature Request] Remove Part Design rejection of nested profiles with >2 levels

adrianinsaval, 2022-02-14:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

starting here https://forum.freecadweb.org/viewtopic.php?p=571084#p570666

### Version

0.20 (Development)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.27428 (Git)
Build type: Release
Branch: master
Hash: 27460358508a2057e0ec57a418641435f12628dd
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Spanish/Paraguay (es_PY)
```

### Subproject(s) affected?

PartDesign

### Idea description

While double nested profiles would result in multiple solids if using on a pad (or other feature) in an empty body, it can very well be used to produce a single solid if there is an existing shape in the body. Currently PD rejects this type of sketch but IMO it is not necessary to reject it, we can see in Part Extrude that his kind of shapes work without issues. See attached example were attempting to pad Sketch001 is rejected but wouldn't lead to multiple solid if accepted and works without issues in Extrude003
[extrudeExample-singleSolid.zip](https://github.com/FreeCAD/FreeCAD/files/8061178/extrudeExample-singleSolid.zip)

Example revolves around pad but this possibly applies to most other PD features.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2022-02-14:
@donovaly

donovaly, 2022-02-14:
The bug is that the sketch is taken, extruded and then it is checked if the result would be several bodies.
To fix this, the extrude result must first be merged to the existing solids and then checked if the result would be several solids.

kisolre, 2022-02-14:
Are you sure about that? As I mentioned in the forum if the hexagon is moved outside the other shape but still over the base  Pad works perfectly fine. And this still creates a Tool which is not a single solid.

donovaly, 2022-02-14:
> As I mentioned in the forum if the hexagon is moved outside the other shape but still over the base Pad works perfectly fine.

I don't understand. Can you please attach your example in the forum thread?

kisolre, 2022-02-14:
> As the code is, the intermediate extrusion is fist made permanent and union with the existing solids after the check of the intermediate extrusion.
If that was true this second pad would also fail:
[image]

donovaly, 2022-02-15:
As I wrote meanwhile in the forum, you are right. The bug is the following:

Part Extrude uses by default Part::FaceMakerBullseye and this face maker can handle the nested structures. PartDesign uses by default Part::FaceMakerCheese.

The issue can be reproduced by changing in the example file for "Extrude003" the Face Maker Class from "Part::FaceMakerBullseye" to "Part::FaceMakerCheese".

I will have a look.

PR #6381 (2022-02-16): [PD] change default FaceMaker to Bullseye
Part does already use this FaceMaker since it can also handle islands.

The PR therefore
- uniforms the behavior of PD and Part
- fixes #6366
- enables to use nested sketches for the various PD features like Revolve

Here is an example file what becomes now possible:
[revolveExample.zip](https://github.com/FreeCAD/FreeCAD/files/8076715/revolveExample.zip)
[image]
commit 7786e24f (2022-02-16): [PD] change default FaceMaker to Bullseye

Part does already use this FaceMaker since it can also handle islands.

The PR therefore
- uniforms the behavior of PD and Part
- fixes #6366
- enables to use nested sketches for the various PD features like Revolve
closed 2022-02-16 by wwmayer