FreeCAD/FreeCAD #6371: [Bug] Calculix run single-threaded if started via "Run Calculix" text button

atorger, 2022-02-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=18&t=66241

### Version

0.20 (Development)

### Full version info

```shell
OS: Debian GNU/Linux bookworm/sid (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.27319 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 67706b1)
Hash: 67706b14df32b91b948952ea2c9350948ba4d4db
Python version: 3.9.10
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

FEM

### Issue description

You can start Calculix either via the small icon, or by double-clicking the solver object in the tree and press the "Run Calculix" button.

These two ways triggers different code paths in the python code. Via the icon, the ccxtools.py/start_ccx() function is run, which sets the OMP_NUM_THREADS variable so that Calculix is run multithreaded. In the other case, task_solver_ccxtools.py/runCalculix() function is run instead, which does not set the OMP_NUM_THREADS variable and thus Calculix runs in a single thread.

I think it would be worthwhile to consolidate the code and only use one start function to avoid different behaviors like this.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-02-14:
Related to to #6370

donovaly, 2022-03-25:
Fixed with PR #6374

closed 2022-03-25 by donovaly