FreeCAD/FreeCAD #6372: [Bug] Dynamically linked library SketcherGui.so is missing a symbol

Developer1970, 2022-02-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=571437#p571437

### Version

0.20 (Development)

### Full version info

```shell
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.27538 (Git)
Build type: Release
Branch: master
Hash: 6cd68bf644ab65e4ecf53bfe2f619c5096d445af
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.0
OCC version: 7.5.2
Locale: English/Canada (en_CA)
```

### Issue description

I install FreeCAD using Ubuntu snap, and I am using the latest development build. I see a missing symbol in the dynamically linked library /snap/freecad-ppd/158/usr/lib/SketcherGui.so as follows:

`Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ImportError'>: /snap/freecad-ppd/158/usr/lib/SketcherGui.so: undefined symbol: _ZN8Sketcher12GeoElementId5RtPntE
Cannot create object 'Body': (/snap/freecad-ppd/158/usr/lib/SketcherGui.so: undefined symbol: _ZN8Sketcher12GeoElementId5RtPntE)
Cannot create object 'Sketch': (/snap/freecad-ppd/158/usr/lib/SketcherGui.so: undefined symbol: _ZN8Sketcher12GeoElementId5RtPntE)
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ImportError'>: /snap/freecad-ppd/158/usr/lib/SketcherGui.so: undefined symbol: _ZN8Sketcher12GeoElementId5RtPntE
Cannot create object 'Revolution': (/snap/freecad-ppd/158/usr/lib/SketcherGui.so: undefined symbol: _ZN8Sketcher12GeoElementId5RtPntE)
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ImportError'>: /snap/freecad-ppd/158/usr/lib/SketcherGui.so: undefined symbol: _ZN8Sketcher12GeoElementId5RtPntE
Cannot create object 'Fillet': (/snap/freecad-ppd/158/usr/lib/SketcherGui.so: undefined symbol: _ZN8Sketcher12GeoElementId5RtPntE)
Cannot create object 'Sketch008': (/snap/freecad-ppd/158/usr/lib/SketcherGui.so: undefined symbol: _ZN8Sketcher12GeoElementId5RtPntE)
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ImportError'>: /snap/freecad-ppd/158/usr/lib/SketcherGui.so: undefined symbol: _ZN8Sketcher12GeoElementId5RtPntE
Cannot create object 'SubtractiveLoft': (/snap/freecad-ppd/158/usr/lib/SketcherGui.so: undefined symbol: _ZN8Sketcher12GeoElementId5RtPntE)
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ImportError'>: /snap/freecad-ppd/158/usr/lib/SketcherGui.so: undefined symbol: _ZN8Sketcher12GeoElementId5RtPntE
Cannot create object 'PolarPattern': (/snap/freecad-ppd/158/usr/lib/SketcherGui.so: undefined symbol: _ZN8Sketcher12GeoElementId5RtPntE)
Cannot create object 'Sketch012': (/snap/freecad-ppd/158/usr/lib/SketcherGui.so: undefined symbol: _ZN8Sketcher12GeoElementId5RtPntE)
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ImportError'>: /snap/freecad-ppd/158/usr/lib/SketcherGui.so: undefined symbol: _ZN8Sketcher12GeoElementId5RtPntE
Cannot create object 'Pad': (/snap/freecad-ppd/158/usr/lib/SketcherGui.so: undefined symbol: _ZN8Sketcher12GeoElementId5RtPntE)
Cannot create object 'Sketch013': (/snap/freecad-ppd/158/usr/lib/SketcherGui.so: undefined symbol: _ZN8Sketcher12GeoElementId5RtPntE)
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ImportError'>: /snap/freecad-ppd/158/usr/lib/SketcherGui.so: undefined symbol: _ZN8Sketcher12GeoElementId5RtPntE
Cannot create object 'Pocket': (/snap/freecad-ppd/158/usr/lib/SketcherGui.so: undefined symbol: _ZN8Sketcher12GeoElementId5RtPntE)
Cannot create object 'Sketch015': (/snap/freecad-ppd/158/usr/lib/SketcherGui.so: undefined symbol: _ZN8Sketcher12GeoElementId5RtPntE)
...
`
This is clearly a missing GeoElementId::RtPnt  function that is not linked in this library.
 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-02-15:
> I am using the latest development build  
Please always post your full [About](https://wiki.freecad.org/About) info

How can one recreate this issue? Just starting FreeCAD makes it crash? or change to the Sketcher workbench? etc... more info pls. In this case I am able to recreate this by starting freecad-ppd 'edge' and switching to Sketcher workbench.

 I get a '_Workbench Failure_' dialog box with the following:
`/snap/freecad-ppd/158/usr/lib/SketcherGui.so: undefined symbol: _ZN8Sketcher12GeoElementId5RtPntE`
[image]

OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.27538 (Git)
Build type: Release
Branch: master
Hash: 6cd68bf644ab65e4ecf53bfe2f619c5096d445af
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.0
OCC version: 7.5.2
Locale: English/United States (en_US)

CC @ppd https://github.com/FreeCAD/FreeCAD/commit/5e5e69013b

ppd, 2022-02-15:
@luzpaz Well, the edge builds will pick up the fix when it lands in master. Snap users can also revert to an older revision: https://snapcraft.io/docs/getting-started#heading--revert until this is fixed.

Developer1970, 2022-02-15:
I see that Ubuntu snap is reverted to rev 157 (commit hash 9b1245ab) from 158, and the issue is gone for now. So, I am going to close this issue for now considering there is a fix for it.

luzpaz, 2022-02-16:
wmayer pushed a fix in https://github.com/FreeCAD/FreeCAD/commit/6e012520d90772c6220e404060046e80525e7108

luzpaz, 2022-02-16:
Should be fixed in
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.27556 (Git)
Build type: Release
Branch: master
Hash: 6e012520d90772c6220e404060046e80525e7108
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.0
OCC version: 7.5.2
Locale: English/United States (en_US)

closed 2022-02-16 by luzpaz