FreeCAD/FreeCAD #6375: [Feature Request] [TechDraw] Surface finish symbols

FEA-eng, 2022-02-15:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=35&t=66071

### Version

0.20 (Development)

### Full version info

```shell
OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.27518 (Git)
Build type: Release
Python version: 3.8.12
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Techdraw

### Idea description

It would be great to have surface finish annotations implemented in TechDraw. These symbols are drawn in a similar manner as in the case of welds - they have a form of modified (depending on the requirement for machining) checkmarks with additional fields for numbers and text. They can describe roughness, lay and waviness but also denote the type of surface treatment. Adding them to TechDraw is necessary because they can't be simply copied from the internet. And since they are similar to weld symbols, maybe it will be easier to implement them.

### Anything else?

A YouTube video showing how to create and import such symbols (could be helpful for the implementation): https://www.youtube.com/watch?v=oCEhTrvF9-M

A picture showing how these symbols are implemented in SolidWorks is attached in the forum thread. There one may also find editable SVG symbol shared by a user to aid the development of this feature.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2022-07-22:
Solved by: #7227

luzpaz, 2022-07-22:
@FEA-eng any idea if this has been added to the documentation?

FEA-eng, 2022-07-22:
> @FEA-eng any idea if this has been added to the documentation?

Yes, it's documented: https://wiki.freecadweb.org/TechDraw_SurfaceFinishSymbol

and I added it to the release notes.