FreeCAD/FreeCAD #6376: [Bug] Placing the working plane grid on a linked part

VictorNine, 2022-02-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=66300

### Version

0.19 (Release)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +99 (Git)
Build type: Release
Branch: Branch_0.19.3
Hash: 6530e364184ce05ccff39501e175cf2237e6ee4b
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
```

### Subproject(s) affected?

Draft

### Issue description

It is not possible to place the working plane on the face of a linked part. The linked part has been moved from the original location. Instead of being placed on the face of the linked part as expected it gets placed on the “original” part. Please see the forum post for details and an example file.

This issue makes it very difficult to use the Dodo workbench on an assembly of linked parts.

This seems to be a bug in Gui.Selection.getSelectionEx()

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-04-15:
There is some activity in the [forum thread](https://forum.freecadweb.org/viewtopic.php?f=3&t=66300)  
CCing @mwganson

Roy-043, 2024-06-13:
As explained in the mentioned forum thread Gui.Selection.getSelectionEx() can handle links as well.
The issue with the Draft WP has been fixed in #11010.

closed 2024-06-13 by Roy-043