FreeCAD/FreeCAD #6395: [Bug] Spreadsheet cell values are not correctly saved / restored

marzn, 2022-02-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/

### Version

0.19 (Release)

### Full version info

```shell
OS: Linux (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24366 (Git)
Build type: Release
Branch: makepkg
Hash: 6530e364184ce05ccff39501e175cf2237e6ee4b
Python version: 3.10.2
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

---

OS: Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.27630 (Git)
Build type: Unknown
Branch: master
Hash: 13fb62fc3f9d65daf5b2b14e250a9a61267d8281
Python 3.10.2, Qt 5.15.2, Coin 4.0.1, OCC 7.5.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Spreadsheet

### Issue description

### Description

If I type `'3M` into a spreadsheet cell, it will be displayed as `3M` which is correct. If I save and reopen the document, it will now display `0.05 °` in the cell.

### How to reproduce

1. Open FreeCAD
2. File -> New
3. Select Workbench Spreadsheet
4. Create new Spreadsheet
5. Click on cell A1
6. Type `'3M`
7. Enter
8. Type `=<<3M>>`
9. Enter
10. Type `=<<%s>> % <<3M>>`
11. File -> Save
12. Close FreeCAD
13. Open previous saved file
14. Open spreadsheet
15. Cell content for A1 and A2 changed to `0.05 °`

### Additional info from python console

After step 10 the python console gives the following values:
```
>>> App.ActiveDocument.Spreadsheet.A1
'3M'
>>> App.ActiveDocument.Spreadsheet.A2
'3M'
>>> App.ActiveDocument.Spreadsheet.A3
'3M'
>>> App.ActiveDocument.Spreadsheet.cells.Content
'<Cells Count="3" xlink="1">\n    <XLinks count="0">\n    </XLinks>\n    <Cell address="A1" content="3M" />\n    <Cell address="A2" content="3M" />\n    <Cell address="A3" content="=&lt;&lt;%s&gt;&gt; % &lt;&lt;3M&gt;&gt;" />\n</Cells>\n'
```

### Document.xml content (extracted from saved file)

```
<!-- snip -->
<Cell address="A1" content="3M" />
<Cell address="A2" content="3M" />
<Cell address="A3" content="=&lt;&lt;%s&gt;&gt; % &lt;&lt;3M&gt;&gt;" />
<!-- snip -->
```

### Expected

If I change the cells in Document.xml to the following:
```
<!-- snip -->
<Cell address="A1" content="'3M" />
<Cell address="A2" content="=&lt;&lt;3M&gt;&gt;" />
<Cell address="A3" content="=&lt;&lt;%s&gt;&gt; % &lt;&lt;3M&gt;&gt;" />
<!-- snip -->
```
and open the document, it will correctly be opened and shows the values in the spreadsheet as before it was closed.

### Possible bug location

I am not sure, may this is completely misleading, but from having a short look at the code, I would expect the bug in `Cell::setContent` in line https://github.com/FreeCAD/FreeCAD/blob/7b7b76f1ec741151efda3af20b219d82a75f6632/src/Mod/Spreadsheet/App/Cell.cpp#L305

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2022-02-18:
Thank you for the report -- we normally do bug triage in the forums, so to that end:
https://forum.freecadweb.org/viewtopic.php?f=3&t=66423

Now we wait to see what others have to report..

bopoh13, 2022-05-27:
```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python 3.8.6+, Qt 5.15.2, Coin 4.0.1, OCC 7.5.0
```
I confirm the problem. After saving and opening spreadsheet the symbol `'` is replaced by the symbol `=`.

<details>
<summary>:arrow_heading_down: And spreadsheet links are duplicated when extracting</summary><br />

https://github.com/lugovoe/FreeCAD-Models/blob/b798137e/PocketTools/BeagleKeychain/Document.xml#L85-L125

</details>

0penBrain, 2023-05-18:
Would that be an acceptable solution that strings are always prefixed with ' (single-quote) when displayed in the "Content" line on top of the spreadsheet ?

adrianinsaval, 2023-05-18:
I personally don't mind, but more casual users might

chennes, 2023-07-20:
This bug is currently tagged for 0.21, but there does not appear to be a working solution on the horizon. @adrianinsaval, @0penBrain do you have a suggestion for a path forward?

adrianinsaval, 2023-07-20:
OpenBrain's PR was working, I think it should be merged and we can adjust the test later

chennes, 2023-07-20:
Are we talking about #9667?

adrianinsaval, 2023-07-20:
Yes

chennes, 2023-07-20:
I don't want to merge a PR that @0penBrain has closed, and I'm not sure exactly what the end state of it was. I see some discussion in the PR, but no overt issues with it. @0penBrain?

0penBrain, 2023-07-20:
The PR was judged to not provide enough tests. I did close as I wasn't able to provide more on this.

0penBrain, 2023-07-23:
I looked at the 0.21 dashboard and saw only a few issues remaining.
Please tell me if one want me to reopen the fixing PR, but be aware I won't do more on it. :smile:

chennes, 2023-07-23:
CONTRIBUTING.md does not require that a PR have 100% test coverage, or anything like that, so I'd be happy to consider the PR in its current state. If more work is required I'm happy to merge the PR as is and let someone else take a stab at it, if that's your preference, @0penBrain.

0penBrain, 2023-07-23:
PR is reopen.
And yes, anybody wanting to increase test coverage is welcome.

commit ee3b777f (2023-07-24): Spreadsheet: fix string save/restore, fixes #6395
closed 2023-07-24 by adrianinsaval