FreeCAD/FreeCAD #6407: [Feature Request] 2d fillet python bindings

looooo, 2022-02-20:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=22&t=49562&p=425165&hilit=ChFi2d#p425165

### Version

0.20 (Development)

### Full version info

```shell
0.20.latest (not at han right now), why is this imortant for feature requests?
```

### Idea description

It would be nice to have python bindings for the occ 2d fillet method `ChFi2d`.

### Anything else?

https://forum.freecadweb.org/viewtopic.php?f=22&t=49562&p=425165&hilit=ChFi2d#p425165

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

looooo, 2022-05-27:
@donovaly @wwmayer This feature is needed in the gear-wb (development version). Currently python-occ is used for this task, but it would be nice if FreeCAD offers a python command to do 2d fillets.

wwmayer, 2022-05-28:
bd853e5ef292a9b0c0f and b07be8767722b4128ea38

closed 2022-05-28 by wwmayer

looooo, 2022-05-28:
wow, this was fast, thanks.

luzpaz, 2022-05-28:
Thank you wmayer!

donovaly, 2022-05-28:
Many thanks!