FreeCAD/FreeCAD #6410: [Bug] Start Workbench shows blank page

ryanthara, 2022-02-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=4&t=66363

### Version

0.19 (Release)

### Full version info

```shell
OS: macOS 10.16
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24366 (Git)
Build type: Release
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: C/Default (C)

and

OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.27518 (Git)
Build type: Release
Python version: 3.9.10
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: C/Default (C)
```

### Issue description

FreeCAD shows an empty Start Workbench page.

[image]

After some left and right clicking around the page gets sensible. You can click invisible objects and get tooltips.

### Anything else?

The issue is reproduce-able on an Intel MacBook Pro and an current ARM iMac with current macOS Monterey version 12.x.

The version of the user.cfg file provided by Chrisb on the Forum thread shows the start page, but installed AdOns or Workbenches are not visible.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

carsten-bruckhaus, 2022-03-07:
Maybe related:
The Start workbench is also blank (or black, rather) for me in @realthunder's LinkStage3 branch in Manjaro KDE.
```
OS: Manjaro Linux (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.111.28429 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: ec814c42d02f15171f9dbb7f650400c9943abfe0
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```
[image]

In version 0.19 (installed from official community repo), Start workbench works as expected.
```
OS: Manjaro Linux (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24366 (Git)
Build type: Release
Branch: makepkg
Hash: 6530e364184ce05ccff39501e175cf2237e6ee4b
Python version: 3.10.2
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

luzpaz, 2022-04-15:
@PLChris @chennes @KeithSloan is this a QtWebEngine issue? Can you still reproduce on v0.20 macOS ?

JohnOCFII, 2022-04-16:
> @PLChris @chennes @KeithSloan is this a QtWebEngine issue? Can you still reproduce on v0.20 macOS ?

I'm not one of those fine individuals, but I still see a blank Start WB with the latest Conda weekly build 28697.  The Start WB seems to work properly with the build from @ipatch (his build 28575 is what I tested).  Need to see if we can integrate his build magic with the Conda builds.

This on macOS Monterey 12.3.1 on an M1 MacBook Pro.

luzpaz, 2022-04-29:
@carsten-bruckhaus still an issue?

carsten-bruckhaus, 2022-04-29:
> @carsten-bruckhaus still an issue?

Yes, still an issue in latest LinkBranch installed via AppImage or snap on Manjaro KDE (snap is auto build and therefor more recent than AppImg) . Still no problem in native FC 0.19, though.

I tried 4 machines, the issue always occured with Qt 5.12 (as bundled in snap and AppImage releases, apparently), never with Qt 5.15.

ipatch, 2022-04-30:
i downloaded the conda release from the weekly releases page, and this is what im seeing...

[image]

JohnOCFII, 2022-05-01:
Still a blank Start workbench for me using the same Conda build (28765).  Perhaps related to macOS 10.16 vs. 10.15?  Or perhaps that I'm on an M1 Mac?

[image]
?

ipatch, 2022-05-01:
start wb wont work on m1 due to qt@5 webengine is not m1 compat nor the qt devs plan on making it compat, so

```
m1 != start wb
```

JohnOCFII, 2022-05-01:
But since the Conda builds are built for Intel, shouldn't this run correctly via Rosetta?  I DO get a Start WB when using your version, as shown below: 

> start wb wont work on m1 due to qt@5 webengine is not m1 compat nor the qt devs plan on making it compat, so
> 
> ```

> m1 != start wb
> ```

[image]

ipatch, 2022-05-01:
okay thanks for that.

m1 does not natively support qt@5 with qt webengine, so if you do see the start wb with homebrew freecad build then its working via rosetta as you stated. however conda uses an older version of qt i believe qt v5.12 or qt v5.13.x where as the versions i publishing use qt v5.15.3 also theres enough difference between the conda builds and hbfc builds for things like this crop up unfortunately.

has the start wb ever worked with a conda release ie. prior release at all?

JohnOCFII, 2022-05-01:
> has the start wb ever worked with a conda release ie. prior release at all?

Good Question.  I pulled down a version of 0.19, and also do NOT see a Start WB.  Last time I used 0.19 I had an Intel Mac.  So -- not sure.  I do think you are on the right track regarding QT v5.12 vs. v5.13 (or something in that area...).

I don't think I've every seen the Start Workbench working in 0.20, though, including when I did have an Intel Mac -- so something MAY have changed to allow that to work, but now that I'm on M1 -- I don't see it.

ageeye, 2022-05-07:
It looks like to locate the problems with webengine in the old qt library. Maybe the Qt 5.15 LTS versions (commercial licenses) fix it. But this is not an option. It remains to switch to Qt 6.
It is possible that Cmake warn if the Qt tree is older than 5.15 and build runs on macOS. 
As we cannot see the wood for the trees, will this warning lose oneself. [See forum](https://forum.freecadweb.org/viewtopic.php?f=4&t=44337#p593737)

ageeye, 2022-05-07:
[Please read this hint](https://forum.freecadweb.org/viewtopic.php?p=593746#p593746) from @looooo

ipatch, 2022-05-12:
https://github.com/FreeCAD/homebrew-freecad/issues/316

donovaly, 2022-05-15:
This bug is also in FC 0.19 -> no blocker for 0.20
The it seems to me the bug only occurs when Qt 5.12 is used. So as solution the MacOS package has to be built using Qt 5.15, right?

andyylin, 2022-06-25:
I also have this issue with 0.20 on Intel MacOS 12.4. Any webpages internal to FreeCAD are blank (Start, Add-on description pages, etc.)

[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: (HEAD detached at 0.20)
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.9.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: C/Default (C)
Installed mods: 
  * Launcher
  * ProDarkThemePreferencePack 1.0.0
  * fcgear 1.0.0
  * Manipulator 1.4.5
  * fasteners 0.3.45
  * Lithophane
  * lattice2 1.0.0
  * Help 1.0.3
  * Curves 0.4.4
  * Assembly3 0.11.3
[/code]

JohnOCFII, 2022-06-25:
Yes, this issue still exists in the Conda build of 0.20.  The Homebrew build does not have this issue.

luzpaz, 2022-06-25:
What does the conda build need ?

donovaly, 2022-06-25:
> Yes, this issue still exists in the Conda build of 0.20. The Homebrew build does not have this issue.

Thanks for this info. So we can write something in our release notes:
https://wiki.freecadweb.org/Release_notes_0.20#Known_Limitations

Can you please make a text proposal for a new entry there about MacOS?

luzpaz, 2022-06-25:
@looooo just a heads up (Conda macOS related)

looooo, 2022-06-25:
For me the start WB works if you switch the software gl option.

luzpaz, 2022-06-25:
> For me the start WB works if you switch the software gl option.  

Care to explain the workaround?

looooo, 2022-06-25:
https://github.com/FreeCAD/FreeCAD/issues/6410#issuecomment-1120241786

looooo, 2022-06-26:
[image]

ryanthara, 2022-06-26:
This helps on macOS Monterey (12.4)

```
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.29126 (Git)
Build type: Release
Branch: master
Hash: 6c1c63cffebe6a9d55bd9b1d8ca6b741a4d16548
Python 3.10.4, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: C/Default (C)
```

luzpaz, 2022-06-26:
Maybe conda can autoset this boolean for macOS builds?

JohnOCFII, 2022-06-27:
> For me the start WB works if you switch the software gl option.

Wow.  Can confirm that choosing "Use software OpenGL" allows the Start Workbench to show up as it should.  Now I need to research what other impact that may have (ie, does it slow down some other graphical performance elsewhere?

[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.29192 (Git)
Build type: Release
Branch: master
Hash: fb1a7af3605ed86631afabb1eb77ae305e418d98
Python 3.10.5, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: C/Default (C)
[/code]

JohnOCFII, 2022-06-27:
Also - as a datapoint, the Homebrew version of 0.20 does not require the `Use software OpenGL` option to be active for the Start Menu to show up normally.

carsten-bruckhaus, 2022-06-27:
`Software OpenGL` does not work with @realthunder Linkstage AppImage version (2022.04.30) in Manjaro KDE -  Start workbench is still blank.

Don't know of this Info is helpful to narrow down the cause of this bug...

adrianinsaval, 2023-05-04:
is this still a problem in the 0.21 dev releases?

JohnOCFII, 2023-05-04:
> is this still a problem in the 0.21 dev releases?

I'm not the OP, but tested and commented above many times.  This is no longer an issue (at least for me) in the 0.21 Dev builds.  
Currently just confirmed with Mac Intel Conda build from a recent weekly, as well as a locally built Mac Arm build using @oursland conda build instructions (since the weekly page isn't yet showing up to date Mac Arm builds).

Intel Build:
```
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.21.0.33040 (Git)
Build type: Release
Branch: master
Hash: 1e830a8800156097ccb3ef6124dced30d4002e76
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.0
Locale: C/Default (C)
```

and 

Mac M1 Arm Build:
```
OS: macOS 13.3
Word size of FreeCAD: 64-bit
Version: 0.21.0.33071 (Git)
Build type: Release
Branch: master
Hash: 703ba9b3cf6f673e6870aa8b7e1badbdeb6c9cd2
Python 3.11.0, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.6.3
Locale: English/United States (en_US)
```

adrianinsaval, 2023-05-04:
Ok, I'll close this for now then, don't hesitate to reopen if the issue is still present for other people

adrianinsaval, 2023-05-04:
You don't have software OpenGL enabled right?

closed 2023-05-04 by adrianinsaval

JohnOCFII, 2023-05-04:
> You don't have software OpenGL enabled right?

Correct - I checked and confirmed that I don't have Software OpenGL enabled.

[image]