FreeCAD/FreeCAD #6413: [Task] Remove remaining Python2 code

luzpaz, 2022-02-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

n/a

### Version

0.20 (Development)

### Full version info

```shell
n/a
```

### Issue description

There is still remaining Python2 related code in the FreeCAD source code. Once a new development version is cut, the task to remove said py2 code should occur very early in the development cycle so any bugs/regressions can be caught and fixed.  

### Anything else?

https://github.com/FreeCAD/FreeCAD/search?q=sys.version_info+%3C+3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sz47, 2022-04-01:
Hello, Ive made a commit on my fork after removing all sys.version_info from the python code. Its in [this](https://github.com/sz47/FreeCAD/commit/dd82d5905b7b93f8558ee22de9daf635181b9a7b) commit. 

If I'm missing something please pardon me, I'd really like to do more. Should I open pull request?

luzpaz, 2022-11-08:
@chennes can we close this ?

chennes, 2022-11-08:
I can easily clean up the last couple of `sys.version_info` checks, but there are still quite a few "py2" search results in the codebase.

PR #7740 (2022-11-09): Remaining py2 removal
Clean up most remaining Python 2 compatibility code. Closes #6413.
closed 2022-11-09 by donovaly