FreeCAD/FreeCAD #6431: [Bug] "gp_Dir::Crossed() - result vector has zero norm"

AndKe, 2022-02-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=573731&sid=d7b328d9216de0f6f0b2a2493035afa4#p573731

### Version

0.20 (Development)

### Full version info

```shell
FreeCAD_weekly-builds-27662-Linux-Conda_glibc2.12-x86_64.AppImage

OS: Ubuntu 21.10 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.27662 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 7a44a5c)
Hash: 7a44a5ca547fa162593b53cc3e017039d99bf4ef
Python 3.9.10, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: English/United Kingdom (en_GB)
```

### Issue description

Threading in reverse direction fails.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2022-02-23:
Confirmed The bug is in the reversed option. When I shift the sketch to be able to omit the Reversed option it works:

[RTK-base0.20-workaround.FCStd](https://forum.freecadweb.org/download/file.php?id=180568)
    (565.28 KiB)

donovaly, 2022-02-23:
@davidosterberg , could you please have a look?

luzpaz, 2022-02-23:
If I can also propose that we improve this error so it offers less generic and technical verbiage ?

wwmayer, 2022-02-26:
Fixed with 4753b0232d