FreeCAD/FreeCAD #6432: [Task] Possibly remove the rest of the 3rd-party CxImage code left in source code

luzpaz, 2022-02-23:
> This is all old code and probably won't compile any more with any newer Windows SDK. IIRC it once was developed for Windows XP but the internal working of thumbnails providers has changed in newer Windows versions. In case it unexpectedly compiles and works with newer Windows versions it should be ported as well and otherwise it can be removed.

> If I don't forget it I will have a look the next time I am on Windows.

Ref: #6388

wwmayer, 2022-02-28:
I haven't expected it but after replacing the CxImage stuff with the Win32 API functions it still compiles.
See 4875aa88b68f0a94

So, can the ticket be closed now?

luzpaz, 2022-02-28:
@wwmayer there are still [remnants](https://github.com/FreeCAD/FreeCAD/search?q=cximage) of `cximage` in the codebase. Is the plan to remove them?

luzpaz, 2022-02-28:
Yet another commit  made by wmayer a3cb945d4e7f525fef62943ac08e6e50dfeafec8

@wmater just one more https://github.com/FreeCAD/FreeCAD/blob/5d49bf78de785a536f941f1a6d06d432582a95d3/src/Tools/LicenseChecker.py#L10

Feel free to add `Closes 6432` to the commit message and it will auto close this ticket :wink:

commit d6c397ea (2022-03-01): Tools: [skip ci] remove the last reference to CxImage: fixes #6432
closed 2022-03-01 by wwmayer