FreeCAD/FreeCAD #6439: [Feature Request] Medical Imagery workbench incl. ability to parse CAD (STL) files out of DICOM files + view them 4 viewports simultaneously

luzpaz, 2022-02-23:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=28362

### Subproject(s) affected?

PartDesign

### Idea description

 Ability to import DICOM files with CAD data (STL/STEP), extract data in to FreeCAD to further modify for whatever purpose. This idea was inspired by multiple forum threads and some social media interactions:

* [How to build a Jaw](https://forum.freecadweb.org/viewtopic.php?f=3&t=28362) (forum thread)
* [mesh to surface, how to do?](https://forum.freecadweb.org/viewtopic.php?f=3&t=39401) (forum thread)
* [stl to solid, works and not, why?](https://forum.freecadweb.org/viewtopic.php?f=3&t=39903) (forum thread)
* https://twitter.com/FreeCADNews/status/1496504162560925698
* [Contributing to Orthanc](https://book.orthanc-server.com/contributing.html?highlight=freecad) (FreeCAD is mentioned on this page)

This ticket has several parts to it:
1. it's related to #5580 which is a feature request to have more than 1 viewports open simultaneously
2. the ability to parse out CAD data from DICOM files (this is also possible through some open source toolchain like 
2a. Use Orthanc server API to pull CAD data out of DICOM and import directly in to FC 
2b. Use workflows discussed on this [page](https://www.researchgate.net/post/Does_anybody_know_any_free_tools_to_convert_a_3D_DICOM_dataset_into_a_surface_representation_file_such_as_STL_for_example) 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-02-24:
CC 
@jodogne ([Orthanc server](https://www.orthanc-server.com/) dev)
@kislinsk ([MITK](http://www.mitk.org/) dev)

Just wanted to keep y'all in the loop on this (feel free to unsubscribe, or I can remove you if you deem that necessary). But we're hoping to collaborate if possible.

luzpaz, 2022-03-04:
@jodogne keep us in the loop, if you don't mind, about orthanc features oriented in the direction of this ticket.

It would be good to know if there are any Blender/Inkscape/Gimp/MeshLab etc... devs that would be interested in collaborating?

luzpaz, 2022-07-16:
@jodogne How's progress on orthanc going ?

jodogne, 2022-07-18:
> @jodogne How's progress on orthanc going ?

I will supervise a student that will (among other) work on the Encapsulated STL IOD during academic year 2022-2023 in the context of his Master's thesis at UCLouvain.

jodogne, 2022-07-18:
> Great news! Any implications on a orthanc->freecad workflow ?

We'll see, this will depend upon the student's progress.

prokoudine, 2023-11-29:
@luzpaz 

> It would be good to know if there are any Blender/Inkscape/Gimp/MeshLab etc... devs that would be interested in collaborating?

There's a patch for GIMP to port the existing plugin to Grassroots DICOM (GDCM) library. But it's from a new contributor, I don't know him at all. And I'm not sure what collaboration would look like. Thoughts?

prokoudine, 2023-11-29:
https://gitlab.gnome.org/GNOME/gimp/-/merge_requests/1135

luzpaz, 2024-04-11:
https://twitter.com/sjodogne/status/1777994287738720634

STL-to-DICOM and DICOM-to-STL plugins available now in Orthanc