FreeCAD/FreeCAD #6445: [Bug] FindPySide2Tools.cmake uses wrong NAMES for uic/rcc

StefanBruens, 2022-02-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=4&t=66440&sid=c846cefba9e6be28ee3ee39bd0a548bc&start=20

### Version

0.20 (Development)

### Full version info

```shell
Fails to build.
```

### Issue description

https://github.com/FreeCAD/FreeCAD/blob/8dd1140a4bf2558472e5c234efc18a017d1f747b/cMake/FindPySide2Tools.cmake#L36-L38

At least Fedora and openSUSE use `uic-qt5`/`rcc-qt5` as binary names, to allow co-installable qt6 tools.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

StefanBruens, 2022-02-24:
Related issue - the AddonManager Mod python part is enabled when BUILD_GUI is ON, but does not declare any dependency on the rcc tool: 
```
[  199s] [  1%] Generating AddonManager_rc.py
[  199s] cd "/home/abuild/rpmbuild/BUILD/FreeCAD-0.20.0~g20220224.86c91d71a1/build/src/Mod/AddonManager" && PYSIDE2RCCBINARY-NOTFOUND --generator=python --compress-algo=zlib --compress=1 /home/abuild/rpmbuild/BUILD/FreeCAD-0.20.0~g20220224.86c91d71a1/src/Mod/AddonManager/Resources/AddonManager.qrc -o /home/abuild/rpmbuild/BUILD/FreeCAD-0.20.0~g20220224.86c91d71a1/build/src/Mod/AddonManager/AddonManager_rc.py
[  199s] /bin/sh: line 1: PYSIDE2RCCBINARY-NOTFOUND: command not found
```

luzpaz, 2022-02-24:
@chennes just a heads-up

chennes, 2022-02-24:
How does one do that? Is it happening automatically in Path, for example, via `qt5_add_resources`?

wwmayer, 2022-02-24:
In your PR you write
`get_property(PYSIDE2UICBINARY TARGET Qt::uic PROPERTY LOCATION)`
and
`get_property(PYSIDE2RCCBINARY TARGET Qt::rcc PROPERTY LOCATION)`

When looking at the file Qt5CoreConfigExtras.cmake or Qt5WidgetsConfigExtras.cmake on my system (which offers Qt 5.9.5) then I find these blocks:
```
    set_target_properties(Qt5::rcc PROPERTIES
        IMPORTED_LOCATION ${imported_location}
    )
...
set(Qt5Core_RCC_EXECUTABLE Qt5::rcc)
```
and
```
    set_target_properties(Qt5::uic PROPERTIES
        IMPORTED_LOCATION ${imported_location}
    )
...
set(Qt5Widgets_UIC_EXECUTABLE Qt5::uic)
```
So, is Qt::rcc and Qt::uic correct or should it rather be Qt5::rcc and Qt5::uic?

chennes, 2022-02-24:
> Related issue - the AddonManager Mod python part is enabled when BUILD_GUI is ON, but does not declare any dependency on the rcc tool:

I just don't know how to tell cMake that the addon manager depends on rcc.

StefanBruens, 2022-02-24:
> So, is Qt::rcc and Qt::uic correct or should it rather be Qt5::rcc and Qt5::uic?

This is the forward compatible notation, which can be used unchanged for both Qt5 and Qt6. As this fragment is guarded by `IF(Qt5Core_VERSION VERSION_LESS 5.14)...ELSE()`, using the new notation is fine.

StefanBruens, 2022-02-24:
> It does it the same way as a couple of other modules like Arch, Draft, ..., Tux. Do you get a failure there, too?

Probably, but the build aborted on first failure.

wwmayer, 2022-02-24:
> This is the forward compatible notation, which can be used unchanged for both Qt5 and Qt6.

OK, I see. 

> Probably, but the build aborted on first failure.

I guess you get the failure because of the differences of [get_property](https://cmake.org/cmake/help/v3.12/command/get_property.html) and [find_program](https://cmake.org/cmake/help/v3.8/command/find_program.html). The latter creates a cache entry so that when using the variable somewhere else its content is used while the former doesn't create a cache entry. When using the variable it probably is empty again.

Maybe it helps if you additionally write
`
set (PYSIDE2RCCBINARY ${PYSIDE2RCCBINARY} CACHE FILEPATH)
`

StefanBruens, 2022-02-24:
> I guess you get the failure because of the differences of [get_property](https://cmake.org/cmake/help/v3.12/command/get_property.html) and [...]

Of course it has to fail when it cant find `rcc` (being due to coding errors as in this case is not the relevant part). Though it *should* fail at cmake run time, not only during build.

If I had run `make` with `-k`, it would have reported/failed on each Mod which depends on rcc.

StefanBruens, 2022-02-24:
> > Related issue - the AddonManager Mod python part is enabled when BUILD_GUI is ON, but does not declare any dependency on the rcc tool:
> 
> I just don't know how to tell cMake that the addon manager depends on rcc.

Simple. Just check for PYSIDE2RCCBINARY in PYSIDE_WRAP_RC.

I have added a commit which does just that: https://github.com/FreeCAD/FreeCAD/pull/6446/commits/7b6decd554d9109d4f03bdf3cfd4a9f2befd4e5f

wwmayer, 2022-02-25:
Does your latest changes fix all issues you had?

StefanBruens, 2022-02-25:
I am not aware of any remaining issues.

PR #6446 (2022-02-26): [CMake] Use imported executable targets for finding Qt uic/rcc
Commit 024bde641c27 ("Update FindPySide2Tools.cmake") removed the
versioned names of uic/rcc, which are at least used on openSUSE
and Fedora.

Instead of determining the path of uic/rcc manually, just use
the imported executable targets. rcc comes from QtCore, while
uic comes from QtWidgets.

Fixes #6445.

- [x]  Your pull request is confined strictly to a single module. That is, all the files changed by your pull request are either in `App`, `Base`, `Gui` or one of the `Mod` subfolders. If you need to make changes in several locat [... cut]
commit 4e9363c7 (2022-02-26): [CMake] Use imported executable targets for finding Qt uic/rcc

Commit 024bde641c27 ("Update FindPySide2Tools.cmake") removed the
versioned names of uic/rcc, which are at least used on openSUSE
and Fedora.

Instead of determining the path of uic/rcc manually, just use
the imported executable targets. rcc comes from QtCore, while
uic comes from QtWidgets.

Fixes #6445.
closed 2022-02-26 by wwmayer