FreeCAD/FreeCAD #6450: [Bug] "Set color per face" task panel: color swatch does not update

Roy-043, 2022-02-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=66619

### Version

0.20 (Development)

### Full version info

```shell
OS: Windows 8.1 (6.3)
Word size of FreeCAD: 64-bit
Version: 0.20.27518 (Git)
Build type: Release
Python version: 3.8.12
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: Dutch/Netherlands (nl_NL)
```

### Subproject(s) affected?

Core

### Issue description

In V0.20 the color swatch in the "Set color per face" task panel does not update when you select a colored face. This does happen in V0.19
[image]
.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EuSouDeus, 2022-03-04:
Hum.. I'm trying to recreate it on the 27422 (Git) and it's working well but only applies the color change after i confirm and deselect the highlighted area.

Could you please update to the latest Git version and check if it still doing the same?!

EuSouDeus, 2022-03-04:
At least on my side is working... the only thing different from yours that i noticed is that you're running windows 8.1 (6.3) while I'm running Windows 10.. not sure but it might be an issue OS related.. unfortunately I only have this laptop don't one another PC with 8.1 on it. Let's hope someone else with the 8.1 version avaliable can give it a try and check if the same issue shows up!

[image]
0

OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.27809 (Git)
Build type: Release
Python 3.8.12, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: Portuguese/Portugal (pt_PT)

Roy-043, 2022-03-04:
You misunderstand the issue. Note that the issue was already confirmed on the forum.

See the attached image for the V0.19 behavior.
[image]

wwmayer, 2022-04-17:
@Roy-043 I have tested it with current master (cfe94b00d343) and for me it works as expected. Does it still fail for you?

Roy-043, 2022-04-17:
This has indeed been fixed now. This issue can be closed. Thanks.
```
OS: Windows 8.1 (6.3)
Word size of FreeCAD: 64-bit
Version: 0.20.28671 (Git)
Build type: Release
Python 3.8.13, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
```

wwmayer, 2022-04-17:
Was fixed with d355b9d9e2f836

closed 2022-04-17 by wwmayer