FreeCAD/FreeCAD #6451: [Bug] Face color of PartDesign Body not exported to STEP

Roy-043, 2022-02-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=66604

### Version

0.20 (Development)

### Full version info

```shell
OS: Windows 8.1 (6.3)
Word size of FreeCAD: 64-bit
Version: 0.20.27518 (Git)
Build type: Release
Python version: 3.8.12
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: Dutch/Netherlands (nl_NL)
```

### Subproject(s) affected?

Core

### Issue description

You can apply a face color to a feature in a PartDesign Body ("Set colors..." context menu option). But this color is not exported to STEP.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tormyvancool, 2022-02-26:
Got same issue.

luzpaz, 2022-02-26:
@tormyvancool this tells us nothing without your full [About](https://wiki.freecad.org/About) info

dubijj, 2022-02-28:
A STEP file also looses colors when exporting to an STL file, probably from the same bug.

About Info
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +99 (Git)
Build type: Release
Branch: Branch_0.19.3
Hash: 6530e364184ce05ccff39501e175cf2237e6ee4b
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

tormyvancool, 2022-02-28:
@luzpaz sorry, please here my about:

OS: Windows 10 Version 2009 <<<< not clue why it tells WIndows 10 since it is Windows 11; but ok perhaps it's fine so.
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +99 (Git)
Build type: Release
Branch: Branch_0.19.3
Hash: 6530e364184ce05ccff39501e175cf2237e6ee4b
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

donovaly, 2022-05-17:
@yorikvanhavre maybe you have time to have a look?

yorikvanhavre, 2022-05-17:
Looking at it now, the culprit is basically here:
https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Import/Gui/AppImportGuiPy.cpp#L364
this findColors() function only considers Parts. We should extend it to also consider other view providers which also have a DiffuseColor property (I suppose the Body does its job properly to populate the DiffuseColor). Looking into it now...

yorikvanhavre, 2022-05-17:
There is still something I don't understand here: `PartDesignGui::VierwProviderBody` is derived from `PartGui::ViewProviderPart`, which itself is derived from `PartGui::ViewProviderPartExt`. So in the above code, `if (vp && vp->isDerivedFrom(PartGui::ViewProviderPartExt::getClassTypeId()))` should work for bodies too...
Maybe then the problem is that the body doesn't properly populate its DiffuseColor

donovaly, 2022-05-18:
@wwmayer , what do you think?

wwmayer, 2022-05-18:
It doesn't look like the STEP export function is broken but the Body object hasn't correctly set the colors. If you open the .FCStd file and check `App.ActiveDocument.Body.ViewObject.DiffuseColor` it only prints a single color value (0.8, 0.8, 0.8) which is the standard grey. For comparison the colors of `App.ActiveDocument.Fillet004.ViewObject.DiffuseColor` (the tip) is set to (0.33, 0.66, 1.0).
`App.ActiveDocument.Fillet002.ViewObject.DiffuseColor` (visible but not the tip) prints a long list of colors.

IMO the correct way should be to copy the diffuseColor property of the tip to the body object. So, if you want to export Fillet002 set it as tip and then run `App.ActiveDocument.Body.ViewObject.DiffuseColor=App.ActiveDocument.Fillet002.ViewObject.DiffuseColor`
Now the exported STEP file has the correct colors.

donovaly, 2022-05-20:
> IMO the correct way should be to copy the diffuseColor property of the tip to the body object.

Thanks.
Could you make a PR?

yorikvanhavre, 2022-05-20:
Let me try to do it today ;) then we ask @wwmayer if I can't :sweat_smile:

yorikvanhavre, 2022-05-20:
Ok I guess I need your help @wwmayer :sweat_smile: 
I added the following code under [ViewProviderBody's updateData()](https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/PartDesign/Gui/ViewProviderBody.cpp#L231):
```
        // update DiffuseColor
        Gui::ViewProvider* vptip = Gui::Application::Instance->getViewProvider(tip);
        if (vptip && vptip->isDerivedFrom(PartGui::ViewProviderPartExt::getClassTypeId())) {
            auto colors = static_cast<PartGui::ViewProviderPartExt*>(vptip)->DiffuseColor.getValues();
            this->DiffuseColor.setValues(colors);
        }
```
But no effect, the DiffuseColor is still standard grey after the tip's DiffuseColor has changed (even after touching it). I guess updateData is not the right place to do it, as some GUI-only mechanism probably takes place afterwards... How would you proceed?

wwmayer, 2022-05-20:
> But no effect, the DiffuseColor is still standard grey after the tip's DiffuseColor has changed (even after touching it).

This happens when just loading the test file. However, when you set the new tip via context-menu the correct colors are set.

> I guess updateData is not the right place to do it, as some GUI-only mechanism probably takes place afterwards...

It is the right place (it should be inside the `if (prop == &body->Tip)`) but because the Body has saved the wrong colors they will be restored after the tip property and thus overrides the correct colors again.

> How would you proceed?

It's up to the user to manually change & restore the tip to correct the colors. When saving the project again the correct colors are written to the file. Or we can touch the tip inside `Body::onDocumentRestored()` to do this automatically.

yorikvanhavre, 2022-05-25:
Thanks @wwmayer !