FreeCAD/FreeCAD #6452: [Bug] EnableSelection has impact on SoMouseButtonEvent

Roy-043, 2022-02-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=10&t=65879

### Version

0.20 (Development)

### Full version info

```shell
OS: Windows 8.1 (6.3)
Word size of FreeCAD: 64-bit
Version: 0.20.27518 (Git)
Build type: Release
Python version: 3.8.12
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: Dutch/Netherlands (nl_NL)
```

### Subproject(s) affected?

Core

### Issue description

The EnableSelection parameter (Edit > Preferences > Display > Colors > Enable selection highlighting) has an impact on SoMouseButtonEvents. If the mouse is over a highlighted object the mouse up event is not detected for some reason.

Simple test:

Expected behavior:

1. EnableSelection=false.
2. Switch to the Draft WB.
3. Change the workingplane to Top with [Draft_SelectPlane](https://wiki.freecadweb.org/Draft_SelectPlane).
4. Create a [Draft_Line](https://wiki.freecadweb.org/Draft_Line).
5. Switch off all Draft snaps except End and Mid.
6. Start the [Draft_CubicBezCurve](https://wiki.freecadweb.org/Draft_CubicBezCurve) command.
7. Click an endpoint of the line and hold down the mouse.
8. Move the mouse over the line until you are a little beyond the midpoint.
9. The line is highlighted.
10. Release the mouse.
11. Result: The curve is tangentially connected to the line. The mouse up event has been detected.
12. Cancel the command.

Faulty behavior:

1. Repeat the test with EnableSelection=true.
2. Result: The curve is not tangentially connected to the line. The mouse up event has not been detected.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-05-03:
@wwmayer can you take a look at this at your convenience?

wwmayer, 2022-05-03:
Wasn't able to reproduce it.

luzpaz, 2022-05-03:
> Wasn't able to reproduce it.

Do you mind adding your About info? Thanks!

luzpaz, 2022-08-27:
@wwmayer on what OS ?

Roy-043, 2022-08-28:
FYI: the issue still exists in this version:
[code]
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of FreeCAD: 64-bit
Version: 0.21.30143 (Git)
Build type: Release
Branch: master
Hash: a72bf218f9917df37e811d7e73a36333282028f5
Python 3.8.13, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
[/code]

0penBrain, 2023-05-17:
OK, I investigated this a bit. The "problem" is indeed the selection mechanism.

Actually it can be summed up very simply : when mouse release event appears above an object, the object is selected and the event is consumed.

I see 2 possibilities to fix:
1.  Upgrade the selection mechanism. For example selection would only happen if both mouse press and release events occur above the same and/or within a defined radius
2. Add to FC new functions so that a feature can temporarily disable (then restore in previous state) the selection mechanism so it is guaranteed to receive all mouse events (unaltered).

Both have (to me) balanced pros and cons so feel free to give your opinion if one looks better to you than the other.

Roy-043, 2023-05-20:
3rd possibility:
It may be possible to implement the 2nd option in the Draft code itself. [Draft_CubicBezCurve](https://wiki.freecad.org/Draft_CubicBezCurve) would then temporarily change the EnableSelection preference.

0penBrain, 2023-05-20:
> 3rd possibility: It may be possible to implement the 2nd option in the Draft code itself. [Draft_CubicBezCurve](https://wiki.freecad.org/Draft_CubicBezCurve) would then temporarily change the EnableSelection preference.

I saw a problem with that. Actually there is an observer on this value that will disable selection for all views if you change it. So one working on different views/documents may see a weird behavior.

Roy-043, 2023-05-20:
What would a user do in those other views while Draft_CubicBezCurve is active?

0penBrain, 2023-05-20:
I don't know. What they want. 😄

Roy-043, 2023-05-21:
An active command will block most other commands.

I am going to have a stab at the 3rd possibility. Implementing either of the other two solutions would probably be too risky for the V0.21 release anyway.

Roy-043, 2023-05-23:
PR #9640 fixes the issue for the Draft_CubicBezCurve command only.

closed 2023-05-24 by Roy-043

0penBrain, 2023-06-08:
@Roy-043 do you know if there are other (Draft) tools that may be concerned and not be fixed ATM?

Roy-043, 2023-06-10:
AFAIK there are no other (Draft) commands that are affected. The only command with dragging that comes to my mind is [Std_BoxSelection](https://wiki.freecad.org/Std_BoxSelection).

A user has reported a problem with the new code (but without a reproducible scenario):
https://forum.freecad.org/viewtopic.php?t=78898

I have tried to reproduce the problem (on Windows and Linux), but have been unable to do so.

PR #9640 (2023-05-24): Draft: Draft_CubicBezCurve: set EnableSelection to False, fixes #6452
While the command is active set EnableSelection to False to correctly detect mouse up events.

Fixes #6452.

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR

---
commit 6835eaf7 (2023-05-24): Draft: Draft_CubicBezCurve: set EnableSelection to False, fixes #6452 (#9640)

While the command is active set EnableSelection to False to correctly detect mouse up events.