FreeCAD/FreeCAD #6453: [vulnerability] <zipios>

mosfet80, 2022-02-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewforum.php?f=4&sid=db4c985f788f36bb46319af246d1b461

### Version

0.20 (Development)

### Full version info

```shell
zipios 0.1.5 into the forder /src/zipios++ is affected to an old vulnerability https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2019-13453 it is necessary update or use only exteral library
```

### Subproject(s) affected?

Other (specify in description)

### Issue description

zipios 0.1.5 into the forder /src/zipios++ is affected to an old vulnerability https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2019-13453 it is necessary update or use only exteral library

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-02-25:
What should we tag these tickets?

AjinkyaDahale, 2022-02-25:
> What should we tag these tickets?

Maybe "3rdParty" and "vulnerability"/"security". It may sound confusing to call it "3rdParty" but any developer who can fix this will know that folder exists and how to fix it.

chennes, 2022-02-25:
For the benefit of anyone reading this: the issue is that a specially crafted zip file may cause zipios to enter into an infinite loop, thus locking up the program. It's categorized as a "denial of service."

wwmayer, 2022-03-10:
Does anybody know of a corrupted zip files that shows the vulnerability of zipios?

FYI, for our zipios version in FreeCAD I fixed this infinite loop in readUint32() 14 years ago and I wanted to see if this fix is sufficient.

closed 2022-03-11 by wwmayer

wwmayer, 2022-03-11:
FreeCAD was not directly affected by this vulnerability because instead of using ZipFile to load an archive it uses the class ZipInputStream. Inside the constructor it calls
`is.exceptions(istream::eofbit | istream::failbit | istream::badbit)`
See also: [exceptions](https://www.cplusplus.com/reference/ios/ios/exceptions/)

This means when the function read() inside readUint32() or readUint16() fails it will throw an exception and thus it's impossible to run into an infinite loop.

So, the issue may show up when using ZipFile but I haven't examined this further. [Here](https://salvatoresecurity.com/fun-with-fuzzers-how-i-discovered-three-vulnerabilities-part-2-of-3/) is a thoroughly explanation of the vulnerability. The central point is to jump at the end of the file if the local header offset is wrong and a subsequent call of readUint32() or readUint16() causes the endless loop.

Nevertheless I have applied the patch from https://sourceforge.net/p/zipios/news/2019/07/version-017-cve-/ so the issue is solved.

luzpaz, 2022-03-11:
@wwmayer is it worth it for us to manage our forked dependency of zipios or can we just use upstream?

wwmayer, 2022-03-12:
Our forked zipios has several improvements over the official version:
* it fixes unicode handling on Windows of the class ZipFile
* we add the export macro to several important classes so that they can be used outside FreeCADBase
* there are several (minor) bugfixes

commit 87f848a2 (2022-03-11): zipios++: Apply patch from https://sourceforge.net/p/zipios/news/2019/07/version-017-cve-/ (fixes #6453)