FreeCAD/FreeCAD #6488: [Bug] Cone surface returns wrong dv derivative

tomate44, 2022-03-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=10&t=66677

### Version

0.20 (Development)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.20.27763 (Git)
Build type: Release
Branch: master
Hash: 75dd3d6533c6eb8a6300da4f4f2ed14b87d54699
Python 3.10.2, Qt 5.15.2, Coin 4.0.1, OCC 7.5.3
Locale: French/France (fr_FR)
Installed mods: 
  * CurvesWB 0.3.0
```

### Subproject(s) affected?

Part

### Issue description

ConeSurface.getDN(u, v, 0, 1) returns a wrong vector that is not tangent to surface.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2022-03-01:
A workaround is implemented but the issue should still be reported to OCC

luzpaz, 2022-03-01:
So far @chennes and @abdullahtahiriyo have been the ones reporting upstream OCC bugs. In order to increase the probability that the OCC team would address tickets is if the submitter can attach a OCC tcl script that replicates the bug.

wwmayer, 2022-03-02:
I opened a ticket there: https://tracker.dev.opencascade.org/view.php?id=32863

luzpaz, 2022-03-02:
@wmayer Thanks! I"ll track this in the OCC Bugs in the Bugtracker [thread](https://forum.freecadweb.org/viewtopic.php?t=20264). Question: why does the ticket specify OCC v7.3.0 ? (shouldn't it be tested on v7.6.1 ?)

wwmayer, 2022-03-02:
> Question: why does the ticket specify OCC v7.3.0 ?

Because I have 7.3.0

luzpaz, 2022-03-20:
Posted to the OCCT forum https://dev.opencascade.org/content/bug-32863-cone-surface-returns-wrong-v-derivative

luzpaz, 2022-05-25:
Good news! Looks like @wwmayer's upstream ticket (https://github.com/FreeCAD/FreeCAD/issues/6488#issuecomment-1057046943) was merged and will be released as part of OCC 7.7 :partying_face:

luzpaz, 2022-07-17:
Related: b02705d79f (by wwmayer)

luzpaz, 2022-11-07:
As mentioned above, Upstream has merged @wwmayer's fix in to OCC 7.7.0. 
Upstream: https://tracker.dev.opencascade.org/view.php?id=32863  
Ref: https://forum.freecadweb.org/viewtopic.php?f=10&t=20264

closed 2022-11-07 by luzpaz