FreeCAD/FreeCAD #6500: [Feature Request] Making format specifier options more clear

FEA-eng, 2022-03-02:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=35&t=65507

### Subproject(s) affected?

Techdraw

### Idea description

The general idea is to make it clear for users which format specifiers can be used for dimensions in TechDraw (especially that one of them is non-standard). There are a few ways to do this:
- move the default format specifier option from Edit --> Preferences --> TechDraw --> Advanded --> Dimension Format to Edit --> Preferences --> TechDraw --> Dimensions
- fill the default format specifier field (it can be confusing when it's empty)
- add a note below the aforementioned field in the Preferences to inform about available formats
- make an expandable list from which formats can be selected
- make a live field where exemplary dimension value can be specified

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2022-03-04:
Some of this implemented by commit 3c34abc0fa / #6504

luzpaz, 2022-06-02:
@FEA-eng FYI, you can also add images to tickets (RMB 'Copy Image > RMB paste in to github).

luzpaz, 2023-11-06:
@FEA-eng what's left to do here after #6504 ?

FEA-eng, 2023-11-06:
> @FEA-eng what's left to do here after #6504 ?

Bullet points 1 and 2 are done. The other ones are completely optional. Perhaps the description in the documentation can be considered sufficient.