FreeCAD/FreeCAD #6505: [Feature Request] Additive "shadow" in Sketch mode.

bhs67, 2022-03-03:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=66714&p=576177#p576177

### Idea description

In the Sketch mode, it would be nice if there were a way to have a large Additive object appear as a shadow so a small Subtractive object (inside the Additive object) is visible.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

snelweg, 2022-03-10:
Works in @realthunder branch in part design, will check if it works in Part wb.

luzpaz, 2022-03-24:
This ticket was hastily opened. It really is a FR for the PartDesign

bhs67, 2022-03-24:
What is FR?

~~~~~~~~~~~~~~~~~~~~~~~~~~~~

On Thu, Mar 24, 2022 at 6:28 PM luzpaz ***@***.***> wrote:

> This ticket was hastily opened. It really is a FR for the PartDesign
>
> —
> Reply to this email directly, view it on GitHub
> <https://github.com/FreeCAD/FreeCAD/issues/6505#issuecomment-1078479552>,
> or unsubscribe
> <https://github.com/notifications/unsubscribe-auth/AEKONNPRJHAADKN2VUUZ2C3VBT3CDANCNFSM5P24X5BQ>
> .
> You are receiving this because you authored the thread.Message ID:
> ***@***.***>
>

luzpaz, 2022-03-24:
Feature request

bhs67, 2022-03-24:
Thanks for clarifying.  Do I need to take any more actions?  Thanks!

~~~~~~~~~~~~~~~~~~~~

On Thu, Mar 24, 2022 at 6:42 PM luzpaz ***@***.***> wrote:
>
> Feature request
>
> —
> Reply to this email directly, view it on GitHub, or unsubscribe.
> You are receiving this because you authored the thread.Message ID: ***@***.***>