FreeCAD/FreeCAD #6514: [Bug] What's This? is not working for some workbenches

donovaly, 2022-03-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

mentioned several times in the forum during the last year

### Version

0.19 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.27915 (Git)
Build type: Release
Branch: master
Hash: a55d1b03fd162ce2217e66cf3734d194a13b456d
Python 3.8.6+, Qt 5.15.2, Coin 4.0.1, OCC 7.5.3
Locale: German/Germany (de_DE)
[/code]
```

### Issue description

As of FC 0.19 and also latest 0.20, the What's this feature does not work for the following WBs:

- Drawing
- Image
- Reverse Engineering
- Points
- Raytracing
- Robot
- Sketcher
- Spreadsheet
- Surface
- Web

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-03-05:
[image]

Related to #5926 regardless of being snap package

Roy-043, 2022-03-05:
@luzpaz: The Help files are not include in the dev version.

Roy-043, 2022-03-05:
See also:
https://forum.freecadweb.org/viewtopic.php?f=10&t=66429

donovaly, 2022-03-05:
> This is the dialog that pops up when I try to use `whatis`

This is related to FC 0.20 and special builds. The bug report here describes what is already not working with our release builds of FC 0.19.3.
Can you please try out FC 0.19.3 and report back if you encounter the same WBs not working as I do or if you see other WBs not working? This helps to find the origin, because maybe it is just a bug in the way I compile the Windows release builds.

Roy-043, 2022-03-05:
If you copy the `doc` folder from V0.18 to V0.19 Std_WhatsThis works for more workbenches. So the problem is (at least in part) due to the compiled Help files.

Test V0.19 with the V0.18 Help files:
I have tested the workbenches from your list and the Std commands.
Std_WhatsThis works for the following:
- Std
- Image
- Points (not all commands)
- Sketcher (not all commands)
- Surface

```
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +99 (Git)
Build type: Release
Branch: Branch_0.19.3
Hash: 6530e364184ce05ccff39501e175cf2237e6ee4b
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Dutch/Netherlands (nl_NL)
```

donovaly, 2022-03-18:
This issue will be resolved by the new Help module. With this PR: #6596 the QCH help system will be abandoned.

yorikvanhavre, 2022-03-18:
Indeed this issue is mostly due to the incomplete Doc package, and should be resolved with the switch to the Help module.

FYI, each command defines its "what's this?" attribute in their getResources() dict, which is basically a page name on the wiki. With the Help module , external addons will also be able to define their "what's this?" attribute, as for ex. a URL pointing to an external wiki/documentation

yorikvanhavre, 2022-03-24:
This should be solved with the merging of #6634 . Please test!

donovaly, 2022-03-24:
I tested this and there are UI issues:

- uninstall the Help addon in case you have it already installed
- create a new file and there a sketch
- use What's This? on a Sketcher WB icon
result:
[image]

So the string argument was not filled.

- click now on the button named "FreeCAD"
result: nothing happens. Also the button name should be more informative

donovaly, 2022-03-24:
As solution, the button should be named "Open AddonManager" and open the addon manager with the Help WB preselected so that the user just have to click "Install"

@chennes , can you please help out here?

donovaly, 2022-03-24:
There is also an Addon Manager issue:

- open the Addon Manager and select the Help WB
- Click on the top right on the button "Install"
result: I don't get it installed but this selection window:
[image]

One issue is that the Help WB will be installed also from newbies and for them "pypandoc" is pure mystery, they just need the Help WB installed. Therefore for the HElp WB, the dependencies should be skipped and installed automatically after the user pressed "Install".

----------

- OK, for now I follow the existing path and select both dependencies and press "Yes"
- I press "Install" again
result: I get the dependencies window again and again with the dependencies unselected

Since the user already selected and gave his OK by pressing Yes, the install should begin.
So in effect I cannot find a way to install the Help WB.

In the console I see also this error message:

```
Traceback (most recent call last):
18:26:13    File "D:\FreeCAD-build\Mod\AddonManager\AddonManager.py", line 1063, in <lambda>
18:26:13      ).clicked.connect(lambda: self.dependency_dialog_yes_clicked(repo))
18:26:13  NameError: name 'repo' is not defined
18:26:27  Traceback (most recent call last):
18:26:27    File "D:\FreeCAD-build\Mod\AddonManager\AddonManager.py", line 1063, in <lambda>
18:26:27      ).clicked.connect(lambda: self.dependency_dialog_yes_clicked(repo))
18:26:27  NameError: name 'repo' is not defined
18:29:55  Traceback (most recent call last):
18:29:55    File "D:\FreeCAD-build\Mod\AddonManager\AddonManager.py", line 1063, in <lambda>
18:29:55      ).clicked.connect(lambda: self.dependency_dialog_yes_clicked(repo))
18:29:55  NameError: name 'repo' is not defined
```

chennes, 2022-03-24:
The bug (error you show above) will be fixed shortly, I'm just waiting for a compile to complete.

Regarding your other suggestion: the long-term plan is to give the Addon Manager an API that would allow external code to do things like instruct it to install the Addon, possibly bypassing the optional dependencies dialog, but that feature does not exist yet. So anything we do in that direction right now is going to be pretty ad hoc. In some ways a larger problem is that on first run, the AM is going to want to create its cache, and that can be quite slow. I have long-terms plans to address that as well, but they aren't going to make 0.20.

chennes, 2022-03-24:
@donovaly that bug is fixed in b8ea740c84ab0544d5715be55ad711ee72b90146.

donovaly, 2022-03-24:
> @donovaly that bug is fixed in [b8ea740](https://github.com/FreeCAD/FreeCAD/commit/b8ea740c84ab0544d5715be55ad711ee72b90146).

Thanks. However, there is still a bug in the Addon Manager:
- I selected to install the dependency "metatdata" and this fails:

[image]

donovaly, 2022-03-24:
Besides this, I think it would help much if in this dialog:

[image]

The button there opens the Addon Manager directly. Could this be done?

chennes, 2022-03-24:
> Thanks. However, there is still a bug in the Addon Manager:

Actually, that's a bug in the dependencies for Help -- "metadata" is wrong. I've submitted a PR to Yorik to fix it.

chennes, 2022-03-24:
> The button there opens the Addon Manager directly. Could this be done?

Yes, and in a few minutes we'll be able to auto-select Help (sorry for the delay, it's a simple change but the library is very busy today).

donovaly, 2022-03-24:
> Actually, that's a bug in the dependencies for Help -- "metadata" is wrong. I've submitted a PR to Yorik to fix it.

I have seen this. But given for another package "metadata" is necessary, I think its installation must be fixed anyway, right?

chennes, 2022-03-24:
I don't know how to fix it: my computer just refuses to install it, either in FreeCAD or just from the command line.

chennes, 2022-03-24:
Actually, now that you mention it, `metadata` shouldn't even be allowed, it's not in the allow list.

chennes, 2022-03-25:
Addressing the first issue: @yorikvanhavre , your dialog can now launch the Addon Manager with the Help addon pre-selected, I added that in fad9d481843d09a27a77bb1652764b686ddd9558. To do it, you need to set the parameter "SelectedAddon":
```
        pref = FreeCAD.ParamGet("User parameter:BaseApp/Preferences/Addons")
        pref.SetString("SelectedAddon", "Help")
        Gui.runCommand('Std_AddonMgr',0)
```
This will launch the Addon Manager, and as soon as the AM has downloaded the list of addons from GitHub, select Help. Caching will continue in the background, but won't block the load of your README, or the installation of the addon.

donovaly, 2022-03-30:
@yorikvanhavre , Ping

yorikvanhavre, 2022-03-30:
Sorry guys, I missed that one... Github notifications are a mess... I merged https://github.com/yorikvanhavre/FreeCAD-Help/pull/7 (not sure it's good to make the markdown module a hard request or not, but lets try) and fixing the "FreeCAD" button. Using the default "OK" button for now, but I'll try to add another button next (Users might still want to close the dialog without installing the Help module, for ex. if they have no internet connection)

donovaly, 2022-03-30:
> Using the default "OK" button for now, 

The idea is this:
- add a button "Open Addon Manager". When users press this, the Addon manageer is opened with the Help WB preselected
- add a button "OK" that just closes the dialog

For the release I find it important to have both buttons since complete newbies cannot have the knowledge to install a new WB just to get the help that later tells them how to use the Addon Manager.

yorikvanhavre, 2022-03-30:
Yes, I'm adding the button now, but in any case they will need to use the addon manager, the button will just open it, not install the addon.

donovaly, 2022-03-30:
> the button will just open it, not install the addon.

That is the plan. but the Help WB should be preselected when the addon manager is opened by the button. According to https://github.com/FreeCAD/FreeCAD/issues/6514#issuecomment-1078524420
this should be possible, right?

donovaly, 2022-03-30:
@chennes , there is still an issue with the Addon manager:
- uninstall the Help WB
- restart FC and open the Addon manager, select there the help WB
- click on Install
result: you are informed that there are optional dependencies. As a newbie you cannot judge what this is abut and will most probably use the default settings and this is not to install markdown.
but then you end up with a non-functional What's This? since you are confronted with a desert of text.

I guess the fix would be:
- the 2 dependencies are not optional but mandatory
- preselect these 2 package
Is this right? Can you please have a look?

yorikvanhavre, 2022-03-30:
> this should be possible, right?

Yes, that's what it does now. Now users only need to press the "install" button.

donovaly, 2022-03-30:
I tested the new button, then I selected to install markdown and pypandoc but also with this, I end up that What's This opens this in my Firefox:
https://raw.githubusercontent.com/FreeCAD/FreeCAD-documentation/main/wiki/PartDesign_ShapeBinder.md

And this looks here like this:
[image]

So there is no formatting at all. The issue would be fixed if it would open our Wiki page:
https://wiki.freecadweb.org/PartDesign_ShapeBinder

donovaly, 2022-03-30:
> This opens this in my Firefox: https://raw.githubusercontent.com/FreeCAD/FreeCAD-documentation/main/wiki/PartDesign_ShapeBinder.md

OK, the bug is that it opens the raw version, not the rendered one. So it must open this:
https://github.com/FreeCAD/FreeCAD-documentation/blob/main/wiki/PartDesign_ShapeBinder.md

By the way, in this particular page the images are broken while for other pages like e.g. this one:
https://github.com/FreeCAD/FreeCAD-documentation/blob/main/wiki/PartDesign_Plane.md
they are correctly displayed.

yorikvanhavre, 2022-03-30:
I just update the Help WB so now we can test different conversions, ex:
```
Help.show("Draft Line",conv="markdown")
Help.show("Draft Line",conv="builtin")
```
that should help us investigating possible problems.

For your other problem, indeed if you use the desktop browser output, the contents are not rendered. This is now fixed with https://github.com/yorikvanhavre/FreeCAD-Help/commit/06e5b4bfab1dd3e9b924218b587bc4beb2815e9e

donovaly, 2022-03-30:
> For your other problem, indeed if you use the desktop browser output, the contents are not rendered. This is now fixed with https://github.com/yorikvanhavre/FreeCAD-Help/commit/06e5b4bfab1dd3e9b924218b587bc4beb2815e9e

Many thanks.
So from my side we can close the bug, if we are sure that when newbies install the help WB and get it in a working state installed by using the default settings in the Addon Manager.
I rely on you. If you say the markdown module can stay to be an optional dependency, then fine.

> For the image problem, that is a conversion fault from the wiki-to-markdown script, there are still some here and there, mostly because of weird formatting on the wiki. I'll look into it

Thanks!

yorikvanhavre, 2022-03-30:
Actually I just noticed another issue, the default opening mode in the preferences is in a MDI tab, not the desktop browser... It is now fixed in https://github.com/yorikvanhavre/FreeCAD-Help/commit/06be3365cb669ed474d81a0d604f635c796e94a2

I'll keep this open a little more to not forget to fix the image issue in https://github.com/FreeCAD/FreeCAD-documentation/blob/main/wiki/PartDesign_Plane.md

donovaly, 2022-03-31:
OK, it seems that markdown is a mandatory dependency for the Help WB because without it, I get on every Wht's this call this:

[image]

> I'll keep this open a little more to not forget to fix the image issue in https://github.com/FreeCAD/FreeCAD-documentation/blob/main/wiki/PartDesign_Plane.md

This page is OK, the bug is in this page:
https://github.com/FreeCAD/FreeCAD-documentation/blob/main/wiki/PartDesign_ShapeBinder.md

yorikvanhavre, 2022-04-06:
That's a pandoc message... Looking into it!

yorikvanhavre, 2022-04-08:
Ok the two pages above are solved (thanks to your recent edits I believe), still looking at the pandoc issue, but it's basically just a warning, it does not prevent the page to render, so it's no blocking

donovaly, 2022-04-13:
> it does not prevent the page to render, so it's no blocking

Fine. Removing the issue from the milestone. It also seems to be an upstream bug.

donovaly, 2022-05-13:
I gave it another test today and meanwhile What's this? works fine for me - also with the items I reported previously here.
-> closing

closed 2022-05-13 by donovaly