FreeCAD/FreeCAD #6516: [Bug] Linked Arrays cause error when placed in draft/arch view

dimitarsp, 2022-03-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=23&t=66824

### Version

0.20 (Development)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.27422 (Git)
Build type: Release
Branch: master
Hash: d938733eaf2c2ce7cb18d1cbb56147185c473530
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)
```

### Subproject(s) affected?

Arch

### Issue description

Create an object. Create a linked array in the from the draft WB. Create a draft/arch view on a techdraw page and the the view on the page fails and outputs the following error:

```
10:09:20  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\dimitar\Documents\_apps\FreeCAD_0.20.27422_Win-LPv12.5.4_vc19-x86-64\Mod\Draft\draftfunctions\svg.py", line 930, in getSVG
    return get_svg(obj,
  File "C:\Users\dimitar\Documents\_apps\FreeCAD_0.20.27422_Win-LPv12.5.4_vc19-x86-64\Mod\Draft\draftfunctions\svg.py", line 434, in get_svg
    svg += get_svg(child,
  File "C:\Users\dimitar\Documents\_apps\FreeCAD_0.20.27422_Win-LPv12.5.4_vc19-x86-64\Mod\Draft\draftfunctions\svg.py", line 832, in get_svg
    fill = utils.get_rgb(vobj.ShapeColor,
<class 'AttributeError'>: 'Gui.ViewProviderLink' object has no attribute 'ShapeColor'
10:09:20  Recompute failed! Please check report view.
```

Seems like the draft view is looking for a property that doesn't exist for linked arrays yet - ShapeColor.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2022-03-10:
I fixed the error for now, but we still probably need a better way to detect colors later on

closed 2022-03-10 by yorikvanhavre

luzpaz, 2022-03-11:
@yorikvanhavre shall we open a ticket to track this 'better way to detect colors' ?

commit dd1b0e0a (2022-03-10): Draft: Fixed error in SVG rendering of LinkedArrays - fixes #6516