FreeCAD/FreeCAD #6532: [Bug] Can't save modified document if backups are disabled, file is deleted instead

adrianinsaval, 2022-03-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

starting here https://forum.freecadweb.org/viewtopic.php?f=3&t=66928&p=577785#p577781

### Version

0.20 (Development)

### Full version info

```shell
OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.27809 (Git)
Build type: Release
Python 3.8.12, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: Spanish/Paraguay (es_PY)
Installed mods: 
  * ArchTextures
  * CurvedShapes 1.0.0
  * Curves 0.3.0
  * Defeaturing
  * fasteners 0.3.38
  * Glass
  * SelectorToolbar
  * Silk -709749264.0.0
```

### Subproject(s) affected?

Core

### Issue description

when unticking `preferences->general->documents->Maximum number of backup files to keep when resaving document ` modified files cannot be saved and the previously existing file is removed, with this error shown:
```
10:18:48  Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'OSError'>: No such file or directory: 'C:/Users/username_edited_out/Downloads/test1.FCStd'
10:19:17  Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'OSError'>: No such file or directory: 'C:/Users/username_edited_out/Downloads/test1.FCStd'
```
Saving a newly created file works

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-03-10:
@adrianinsaval to automagically close a ticket from a commit, use `Fixes #<ticket_number>` or `Closes #<ticket_number>`  in the commit summary body.

adrianinsaval, 2022-03-10:
Commit already says `[Fixes #6532]` (it's at the end so you need to expand to see it) is something wrong with how I wrote it?
Of course I imagine it needs to be merged before the issue is automatically closed.

commit c346f76a (2022-03-28): move tmp file rename after file deletion in applyTimeStamp when numberOfFiles <=0 [fixes #6532]
closed 2022-03-28 by wwmayer