FreeCAD/FreeCAD #6533: [Feature Request] <Expose custom pitch and diameter of threaded hole>

acominelli, 2022-03-08:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=19&t=66805&p=577475#p577475

### Subproject(s) affected?

PartDesign

### Idea description

Starting from release 0.20, with the hole tool it is possibile to model real threads.
It would be useful to completely customize the thread, exposing editable pitch and diameter of the thread in the hole menu.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-03-10:
CC @davidosterberg perhaps you're interested in this?

simonachmueller, 2022-08-07:
Hi, I would be also interested in that, wanted to design a custom adapter/nozzle for the PET bottle water rocket for 3D printing.

rjordans, 2023-02-10:
+1 from my side.  I need to model a part that threads onto a camera lens which uses a 49mm x 0.75mm pitch thread and it would be great to visually confirm the dimensions and have them properly specified in the drawings.

petergalgano, 2023-03-12:
This is something I've stumbled on as well, I require modeling a thread of M30x2.5, which isn't a standard, but apparently isn't uncommon to have other pitches, especially on larger diameters, or the situation mentioned above. I went looking through the data, and found json data files for different pockets such as cap head or counterbore. I figured that there would be, but did not find data files for the thread pattern libraries themselves, which I assumed would have drill, inner, outer diameters, fit information, etc. Perhaps this woul

I wouldn't have ability to code such changes, but it occurs to me that it might be a relatively simple step to expose data as json files as a step prior to creating UI changes to allow customization on the fly, which I assume would be quite a bit more work. Custom libraries could be created and shared by users. I'd be happy to help with researching and documenting appropriate values.

Here's a list of extended metric values for what it's worth, and there seem to be reliable standards and documents available on the web to assist with documentation. http://burovik.biz/Metricthread.pdf,

cbabos, 2023-05-12:
This feature would be very appreciated. For 3D printing it would be very useful to customise the thread in the hole. I'm designing currently a filament holder where I want a nut around a custom 30mm thread bolt with 2mm pitch to have a strong and usable plastic thread for fastening the filament holder, and also to strengthen the connection between the spool and the rolling rod.

Anthony-Gaudino, 2023-09-18:
Sorry for spamming, but I also think this feature would be very appreciated.
I have to create multiple threads that are not standard and having this feature would make it much easier.

benj5378, 2024-09-30:
Currently working on this

antekone, 2024-11-05:
@benj5378, do you have partial implementation available somewhere in some fork/branch?

benj5378, 2024-11-05:
@antekone I do have partial implementation. Are you interested in using the feature or doing development?

antekone, 2024-11-05:
@benj5378 using :), but I'm also interested in seeing how such a feature would be implemented if I'd want to do it myself. (never did any development in any CAD though, so it's mostly curiosity)

benj5378, 2024-11-05:
@antekone, I can send it once I get home. The backend is already implemented in main branch. It's only GUI to expose the settings, that hasn't been made.

botikori, 2025-02-01:
@benj5378 Is this feature going to be added in the FC 1.1 release version? I'm looking forward to using it, it would be so helpful for 3D printing, the default ISO profiles are fine, but don't work well for 3D printed threads. Custom pitch and custom diameter would be a game changer.