FreeCAD/FreeCAD #6538: [crash] in making 2D offset of a circle

donovaly, 2022-03-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=577963#p577963

### Version

0.20 (Development)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.27990 (Git)
Build type: Release
Branch: master
Hash: c596c2fe3de933a00ba4245931e8e7bc3093832d
Python 3.8.6+, Qt 5.15.2, Coin 4.0.1, OCC 7.5.3
Locale: German/Germany (de_DE)
Installed mods:
```

### Subproject(s) affected?

Part

### Issue description

- take this simple file that contains a circle in a sketch: https://forum.freecadweb.org/download/file.php?id=182301
- create a 2D offset using the part WB and use "-0.1 mm" as offset
result: as soon as you entered "-0.1" anf then click out of the line edit of the task dialog FC crashes with illegal storage acces in "BRepOffsetAPI_MakeOffset"

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xxxajk, 2022-03-10:
Also happens in the Linux version.
See https://github.com/FreeCAD/FreeCAD-Bundle/issues/95 for full report.

xxxajk, 2022-03-10:
Even though I'm not familiar with the code base at all, I at least know where the crash occurs for me, but not the why.
The crash certainly occurs at https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Part/App/TopoShape.cpp#L2955
Is there anything I can do to help resolve this, such as instrumentation, gdb, etc?

luzpaz, 2022-03-11:
@xxxajk please participate on the designated [forum](https://forum.freecadweb.org/viewtopic.php?p=577963#p577963) thread. Thx!

xxxajk, 2022-03-12:
I did, left some GDB data there. This has to be an OCCT bug.

xxxajk, 2022-03-17:
This is a major show stopper in many cases. I'm kind of surprised that this wouldn't get more rapid care, or at least identify if it is indeed an OCCT bug. If that's the case, does it also affect arcs? I have not had the chance to test that case.

donovaly, 2022-03-17:
The problem is that we cannot do much if it is an OCC bug. We reported many bugs to the OCC guys, even patches but they don't care much. I mean there are for example patches around for more than one year and there were not considered for OCC 7.6.

xxxajk, 2022-03-18:
I understand that, but knowing what the issue is can render a work-around. 
Problem is that I don't understand the problem myself, else I would have offered a (potential) work-around.

0penBrain, 2022-03-28:
I had a look at this. It segfaults deep in OCC and there is probably nothing we can do.
FC already implements a signal catcher but it's not enough. It catches the first segfault but stack or another part of memory stays rubbish and as soon as the function is called a second time it definitely crashes.
@donovaly Does Microsoft SEH succeed handling this one smoothly ?

xxxajk, 2022-03-28:
For now I have resorted to doing some python code to do what I need specificlly, however it's not for normal/regular use.

That said...

If it is deep inside OOC, there's a total possibility to do the size changes in python, and generate a new or replacement toposhape. It might be a bit of work to re-implement, but is totally possible.

donovaly, 2022-03-28:
> Does Microsoft SEH succeed handling this one smoothly ?

What do you mean?
I compile with SEH set on but get the crash when creating a negative offset for a circle sketch.

0penBrain, 2022-03-29:
@donovaly that was my question. If a Windows binary compiled with SEH crashes, there are very few chances that we can recover from the segfault. So probably useless spending time trying to catch the fault in FreeCAD. ;)

xxxajk, 2022-03-29:
...and why I suggest writing it in Python ;-)

donovaly, 2022-03-30:
> ...and why I suggest writing it in Python ;-)

The user can already draw a new circle. So there is a workaround.

However, nothing we can do not -> removing it from the milestone

0penBrain, 2022-03-30:
@donovaly one thing we can probably do as we catch the first segfault is warn the user and advise to cancel in-progress operation, save open documents and restart FC. Is it worth?

donovaly, 2022-03-30:
>Is it worth?

If we can avoid a crash, then yes. But I cannot judge before seeing it in action.

0penBrain, 2022-03-30:
> If we can avoid a crash, then yes. But I cannot judge before seeing it in action.

In this case crash appears only on the 2nd call of the function, so yes, if user stops after 1st call, probably we can avoid the crash.
This said, AFAIK this only concerns Linux OS. Maybe we should think to a wider solution.

donovaly, 2022-03-30:
> so yes, if user stops after 1st call, probably we can avoid the crash.

This would be great. Even when it "only" cures the issue for Linux, a crash prevention is welcome.

0penBrain, 2022-03-30:
> This would be great. Even when it "only" cures the issue for Linux, a crash prevention is welcome.

I did a PoC and admittedly, this is a complete failure. :)

xxxajk, 2022-04-01:
> In this case crash appears only on the 2nd call of the function, so yes, if user stops after 1st call, probably we can avoid the crash. This said, AFAIK this only concerns Linux OS. Maybe we should [...]

That's because the first call applies the changes to the exterior wires of the toposhape. The second call does any internal holes. 
I'll try later to see if it affects a circle as the external wire.
Something else that I have not tested is an arc of a circle...

FEA-eng, 2022-04-03:
> The problem is that we cannot do much if it is an OCC bug. We reported many bugs to the OCC guys, even patches but they don't care much. I mean there are for example patches around for more than one year and there were not considered for OCC 7.6.

Still, maybe we could report this OCC bug. There's always a chance that they will fix this one.

luzpaz, 2022-04-03:
Please x-post to OCC bug to this [thread](https://forum.freecadweb.org/viewtopic.php?t=20264) so we can discuss in more depth. Thanks!

Roy-043, 2022-11-27:
FWIW Python workaround:
```
import Part

offset = -0.3
circle = Part.Circle().toShape()
circle.Orientation = "Reversed"
circle = Part.Wire(circle)
shrunk_circle = circle.makeOffset2D(-offset)

Part.show(circle, "circle")
Part.show(shrunk_circle, "shrunk_circle")
```

xxxajk, 2022-11-30:
Only thought now is to figure out how the "work around" can be implemented within the resize function, or I could preprocess the wires if a circle would end up a bug tripper. 
I already do some processing anyway, but the idea I originally had just used OCC to do that sort of thing...
I don't really understand why it would fault -- totally doesn't make any sense whatsoever.

bgbsww, 2023-12-04:
Here's a brute force fix that basically takes @Roy-043 's workaround, and makes it happen in the makeOffset2D code.  I don't know if this should proceed, or we just wait for the eventual upstream fix.
https://github.com/bgbsww/FreeCAD/tree/bgbsww-patch-17

xxxajk, 2023-12-05:
Personally, for what I needed to do 2D stuff, I just convert to/from shapely, which humorously is faster than OCC, since it can leverage all 24 CPUs here. I was able to process 10,000, complex edge polygons (some containing 1000 edges) in under 3 seconds instead of the ridiculous 3 to 4 hours that it took OCC to do the same exact thing, and that included the conversion between the formats.

Syres916, 2024-10-11:
I don't think this is an issue using OCCT 7.6.3 and 7.8.x (I'll ignore 7.7.x due to the number of known issues). Here is the Python Console output if a unit test needs to be written:

```
>>> App.newDocument()
>>> # App.setActiveDocument("Unnamed")
>>> # App.ActiveDocument=App.getDocument("Unnamed")
>>> # Gui.ActiveDocument=Gui.getDocument("Unnamed")
>>> # Gui.activeDocument().activeView().viewDefaultOrientation()
>>> ### End command Std_New
>>> # Gui.runCommand('Std_OrthographicCamera',1)
>>> ### Begin command Sketcher_NewSketch
>>> App.activeDocument().addObject('Sketcher::SketchObject', 'Sketch')
>>> App.activeDocument().Sketch.Placement = App.Placement(App.Vector(0.000000, 0.000000, 0.000000), App.Rotation(0.000000, 0.000000, 0.000000, 1.000000))
>>> App.activeDocument().Sketch.MapMode = "Deactivated"
>>> # Gui.activeDocument().setEdit('Sketch')
>>> # import Show
>>> # ActiveSketch = App.getDocument('Unnamed').getObject('Sketch')
>>> # tv = Show.TempoVis(App.ActiveDocument, tag= ActiveSketch.ViewObject.TypeId)
>>> # ActiveSketch.ViewObject.TempoVis = tv
>>> # if ActiveSketch.ViewObject.EditingWorkbench:
>>> #   tv.activateWorkbench(ActiveSketch.ViewObject.EditingWorkbench)
>>> # if ActiveSketch.ViewObject.HideDependent:
>>> #   tv.hide(tv.get_all_dependent(App.getDocument('Unnamed').getObject('Sketch'), ''))
>>> # if ActiveSketch.ViewObject.ShowSupport:
>>> #   tv.show([ref[0] for ref in ActiveSketch.AttachmentSupport if not ref[0].isDerivedFrom("PartDesign::Plane")])
>>> # if ActiveSketch.ViewObject.ShowLinks:
>>> #   tv.show([ref[0] for ref in ActiveSketch.ExternalGeometry])
>>> # tv.sketchClipPlane(ActiveSketch, ActiveSketch.ViewObject.SectionView)
>>> # tv.hide(ActiveSketch)
>>> # del(tv)
>>> # del(ActiveSketch)
>>> # 
>>> # ActiveSketch = App.getDocument('Unnamed').getObject('Sketch')
>>> # if ActiveSketch.ViewObject.RestoreCamera:
>>> #   ActiveSketch.ViewObject.TempoVis.saveCamera()
>>> #   if ActiveSketch.ViewObject.ForceOrtho:
>>> #     ActiveSketch.ViewObject.Document.ActiveView.setCameraType('Orthographic')
>>> # 
>>> ### End command Sketcher_NewSketch
>>> # Gui.runCommand('Sketcher_CompCreateConic',0)
>>> lastGeoId = len(ActiveSketch.Geometry)
>>> 
>>> geoList = []
>>> geoList.append(Part.Circle(App.Vector(0.000000, 0.000000, 0.000000), App.Vector(0.000000, 0.000000, 1.000000), 39.805225))
>>> App.getDocument('Unnamed').getObject('Sketch').addGeometry(geoList,False)
>>> del geoList
>>> 
>>> constraintList = []
>>> App.getDocument('Unnamed').getObject('Sketch').addConstraint(Sketcher.Constraint('Coincident', 0, 3, -1, 1))
>>> 
>>> 
>>> # Gui.getDocument('Unnamed').resetEdit()
>>> App.ActiveDocument.recompute()
>>> # ActiveSketch = App.getDocument('Unnamed').getObject('Sketch')
>>> # tv = ActiveSketch.ViewObject.TempoVis
>>> # if tv:
>>> #   tv.restore()
>>> # ActiveSketch.ViewObject.TempoVis = None
>>> # del(tv)
>>> # del(ActiveSketch)
>>> # 
>>> # Gui.Selection.addSelection('Unnamed','Sketch')
>>> App.getDocument('Unnamed').recompute()
>>> ### Begin command Std_Workbench
>>> # Gui.activateWorkbench("PartWorkbench")
>>> ### End command Std_Workbench
>>> ### Begin command Part_Offset2D
>>> App.ActiveDocument.addObject("Part::Offset2D","Offset2D")
>>> App.ActiveDocument.Offset2D.Source = App.ActiveDocument.Sketch
>>> App.ActiveDocument.Offset2D.Value = 1.0
[... 42 lines cut ...]
  * A2plus 0.4.68
```

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38945 (Git)
Build type: Release
Branch: main
Hash: 82a98d6e31659e0f2754bf46fd1ece79430bbe8b
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk 9.1.0, OCC 7.8.2.dev
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Curves 0.6.48
  * A2plus 0.4.68
```

jdegenstein, 2025-08-13:
Possibly related to https://github.com/Open-Cascade-SAS/OCCT/issues/191

stefano2734, 2026-09-12:
> Possibly related to [Open-Cascade-SAS/OCCT#191](https://github.com/Open-Cascade-SAS/OCCT/issues/191)

After a year, is this actual with occt 8.0.1 and 26.3 dev?
Perhaps it is solved or improved.

xxxajk, 2026-09-19:
> > Possibly related to [Open-Cascade-SAS/OCCT#191](https://github.com/Open-Cascade-SAS/OCCT/issues/191)
> 
> After a year, is this actual with occt 8.0.1 and 26.3 dev? Perhaps it is solved or improved.

Doubt it? The performance due to python's GIL is abysmal regardless.

pieterhijma, 2026-10-27:
I cannot reproduce on FreeCAD `main` with OCCT 8.1 dev. Negative offsets and in particular -0.1 work as expected.

```
OS: Arch Linux (tty)
Architecture: x86_64
Version: 27.1.0dev.49321 +1 (Git)
Build date: 2026/10/27 12:27:31
Build type: Debug
Branch: adapt-to-occt-v8_1_0
Hash: 371a550ee620d900a88d2e96bf8cb9e47b7e45ec
Python 3.14.8, Qt 6.12.0, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.1, boost 1_92, Eigen3 5.0.1, PySide 6.12.0
shiboken 6.12.0, SMESH 7.7.1.0, xerces-c 3.3.0, Clipper2 2.0.1, OCC 8.1.0.dev1-53706f78a8
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
OpenGL version/vendor/renderer: 4.6 (Compatibility Profile) Mesa 26.2.5-arch1.1/AMD/AMD Radeon 780M Graphics (radeonsi, phoenix, ACO, DRM 3.64, 7.2.11-arch1-1)
```

I'm assuming that this issue has been resolved. Since it has also been reported to work on 7.6.3 and 7.8, I'm marking this as Done in the OCCT Liaison project and I'm closing this issue.

closed 2026-10-27 by pieterhijma

FEA-eng, 2026-10-27:
> I cannot reproduce on FreeCAD `main` with OCCT 8.1 dev. Negative offsets and in particular -0.1 work as expected.

For me, it fails with:

`19:54:30  <Exception> TopoShapeExpansion.cpp(2880): BRepOffsetAPI_MakeOffset has crashed! (Unknown exception caught)`

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 621.01/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```

pieterhijma, 2026-11-06:
Ok, the version difference is too close for comfort. I will check it out.