FreeCAD/FreeCAD #6545: Constraint symbols seem to be missing on zoom in

PLChris, 2022-03-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=66919

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.27936 (Git)
Build type: Release
Python 3.9.10, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: C/Default (C)
Installed mods: 
[/code]
```

### Subproject(s) affected?

Sketcher

### Issue description

On zooming out the constraint symbols can be grouped. On zooming in again this grouping is not undone in all cases. It needs a hover of the mouse over a geometric element or an axis to regroup which then shows all the constraints.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-07-03:
Hi @abdullahtahiriyo was this issues fixed by #6578 ?

abdullahtahiriyo, 2022-07-06:
@luzpaz 
I would say it was.

closed 2022-07-06 by luzpaz