FreeCAD/FreeCAD #6582: [Bug] [TD]Bitmap hatching is not supported

WandererFan, 2022-03-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://www.forum.freecadweb.org/viewtopic.php?f=35&t=67063

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.
Build type: Release
Branch: unknown
Hash: bda12907dee4fe3f0e7dec8f7e09e098fc5036e4
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, OCC 7.5.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.54b
  * Curves 0.3.0
[/code]
```

### Subproject(s) affected?

Techdraw

### Issue description

Face hatching should allow use of bitmaps as texture fill.  There is code in QGIFace to paint the faces with a bitmap, but the hatch dialog does not create bitmap hatches and QGIViewPart is not aware of bitmap hatching. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AjinkyaDahale, 2022-04-29:
> No, it's a bug. Bitmap fills are supposed to work, but don't.

@WandererFan I saw that only SVGs are supported, and "bitmap" was a raster format. WIth "bitmap" did you mean custom pattern in general? If so I'm sorry.

WandererFan, 2022-04-29:
Faces are supposed to support an SVG tile fill, Autodesk PAT fill, solid
colour or jpeg/png/bmp fill.  The jpeg/png/bmp part is only 1/2 done and
will be fixed in v0.21.

donovaly, 2022-07-03:
fixed by #7121

also backported for 0.20.1

closed 2022-07-03 by donovaly