FreeCAD/FreeCAD #6584: [Bug] Pocket becomes a pad when next object ofter a polar pattern

donovaly, 2022-03-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=579381#p579381

### Version

0.19 (Release)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +148 (Git)
Build type: Release
Branch: Branch_0.19.4
Hash: 476ecf091941bead59b14e44afa6064d5a66afa3
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

PartDesign

### Issue description

See https://forum.freecadweb.org/viewtopic.php?p=579381#p579381 for a description and example file.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2022-03-14:
@realthunder , since you did much to PD in your branch, maybe you have an idea what is causing this nasty issue?

0penBrain, 2022-03-16:
I had a quick look but can't find anything obvious in the code that would create this. I'm balanced between a FC bug and an OCC one...

donovaly, 2022-03-21:
> Caused by [aa19f8b](https://github.com/FreeCAD/FreeCAD/commit/aa19f8b05fbbd2f40f075fcad0e6d22b3dc2bff0)

Since this commit does not seem the be the origin, just a symptom, could it be that this commit causes this bug?: https://github.com/FreeCAD/FreeCAD/commit/03da9d3501

donovaly, 2022-03-21:
> could it be that this commit causes this bug?: https://github.com/FreeCAD/FreeCAD/commit/03da9d3501

No, reverting this commit does not fix the issue.

donovaly, 2022-03-21:
Also reverting https://github.com/FreeCAD/FreeCAD/commit/aa19f8b05fbbd2f40f075fcad0e6d22b3dc2bff0 does not fix the issue.

0penBrain, 2022-03-31:
@AjinkyaDahale Could you clarify about difference between 6665 and 6671 ? What are pros & cons of each ? Is one covering more cases ?

AjinkyaDahale, 2022-04-04:
> @AjinkyaDahale Could you clarify about difference between 6665 and 6671 ?

Just where the inversion check happens. A user there did some benchmarks and found a significant speed difference in #6665. HOWEVER, they have since found an error in that test so now it looks quite better.

> What are pros & cons of each ?

I guess #6671 does the check at the very end, so it happens even if refine is set to false. #6665 on the other hand will also correct the loose solids if that is wanted.

commit 35913fc6 (2022-04-10): Part: fixes #6584: Pocket becomes a pad when next object ofter a polar pattern
closed 2022-04-10 by wwmayer