FreeCAD/FreeCAD #6587: [Feature Request] [TechDraw] Ordinate dimensions

FEA-eng, 2022-03-14:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=35&t=67127

### Subproject(s) affected?

Techdraw

### Idea description

Ordinate dimensions (set of dimensions measured from zero ordinate) could be useful in some cases, e.g. for selected machining processes.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-06:
@FEA-eng can you ping 'kemes' on the forum and ask for the development branch they were working on so we can add it to this ticket ?

FEA-eng, 2023-11-06:
> @FEA-eng can you ping 'kemes' on the forum and ask for the development branch they were working on so we can add it to this ticket ?

Here's a link to that branch (from the forum thread): https://github.com/kemes/FreeCAD_TechDraw_Ordinate_Dimensions/tree/techdraw_ordinate_dimensions

luzpaz, 2023-11-07:
CC @WandererFan

FEA-eng, 2024-05-14:
@PaddleStroke This is one of the most important and commonly requested missing features in TechDraw. Maybe you could implement it for your smart dimension tool at some point? Of course, not necessarily for the upcoming release. As you can see, an attempt has already been made to add this but it was abandoned eventually.

PaddleStroke, 2024-05-15:
Can you explain the difference with the current coordinate dimension? A screenshot will do

FEA-eng, 2024-05-15:
> Can you explain the difference with the current coordinate dimension? A screenshot will do

Coordinate dimension tool creates normally looking dimensions and just involves some automation (you select multiple vertices and it creates several length dimensions starting from the same baseline):

[image]

Ordinate dimensions have a totally different style, with just a single line and value:

[image]

source: https://help.autodesk.com/view/ACD/2024/ENU/?guid=GUID-A2513269-161C-4193-BBC4-1D6717C41A47

PaddleStroke, 2024-05-15:
Yes but technically they are the same right? It's only the style that is different?

FEA-eng, 2024-05-15:
> Yes but technically they are the same right? It's only the style that is different?

Good question. Conceptually, it should be pretty much the same as you select a point from which all the dimensions are created.

[image]

Here we would just need to add the 0 dimension for the left hole and change the style of the existing dimensions to match what is shown in the referenced AutoCAD documentation.

Ordinate dimensions can be vertical or horizontal.

FEA-eng, 2024-10-08:
@FreeCAD/cad-advisory-group It's a standard feature in other CAD software and commonly requested for FreeCAD (in fact, one of the most common FRs for TD). Maybe deserves a place on your list?

pierreporte, 2024-10-08:
It’s tagged both https://github.com/FreeCAD/FreeCAD/labels/Topic%3A%20User%20Interface and https://github.com/FreeCAD/FreeCAD/labels/Type%3A%20Feature. I’ve ignored all https://github.com/FreeCAD/FreeCAD/labels/Topic%3A%20User%20Interface issues. It’s definitely going to the board.

bguest, 2025-02-02:
I picked up "kemes" branch rebased it and started working on in here: https://github.com/bguest/FreeCAD/tree/bg/ordinate-dimensions, this is still very much a draft. I'm still very new to this codebase, so if I'm horrendously off base somewhere please let me know. I still need to:

- figure out how to display origin / zero point. 
- support ISO dimension types
- update widget svg's to new style.
- probably refactoring / cleanup

[image]

pierreporte, 2025-04-29:
A **150 € reward** will be given to the contributor that writes a pull request, solving this issue, that gets merged. It will be paid by the FreeCAD Project Association. It is to be considered an incentive. Grant applications are welcome, but the total amount will exclude the reward.

imoldovavan, 2025-05-28:
https://archive.org/details/asme-y-14.5-2018-dimensioning-and-tolerancing/page/30/mode/2up

jon-durand, 2026-06-28:
@bguest were you still working on this issue? Curious to know the likelyhood of this being merged in the next Freecad release.

pythonyx135793, 2026-09-13:
Hello — this issue is open and unassigned, and PR #32025 was closed without merging. Is the €150 FPA reward stated above still active, and is the issue currently available for a new contributor to implement? I will avoid overlapping work.