FreeCAD/FreeCAD #6588: [Feature Request] Always place the Transform tool's manipulator in the center of a selected body

FEA-eng, 2022-03-15:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=67141

### Subproject(s) affected?

Core

### Idea description

Currently, the Transform tool sometimes places its manipulator far away from the selected body which makes it difficult to move the said body. Ideally, the tool should be placed in the center (possibly COG) of the object being moved.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

macdroid53, 2022-06-13:
https://github.com/macdroid53/FreeCAD-Tips-n-Workflows#dealing-with-lost-in-space-import-geometry

macdroid53, 2024-02-24:
The transform tool should not be inside the selected object (I specifically don't use the term Body, because the tool is used on more than just Part Design Body objects).