FreeCAD/FreeCAD #6593: [Bug] Fail to load a project with arbitrarily named constraints in a sketch

wwmayer, 2022-03-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=578773

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 18.04.6 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.28123 (Git)
Build type: Debug
Branch: master
Hash: 0347f6f91b82dc783d7902a8c58bafa00fa77ae9
Python 3.6.9, Qt 5.9.5, Coin 4.0.0a, OCC 7.3.0
Locale: German/Germany (de_DE)
[/code]
```

### Subproject(s) affected?

Core

### Issue description

The name for a constraint of a sketch allow any character which however the expression cannot handle. So, a name can be e.g. `Bottom, height`. When saving and loading such a project file the expression parser will get this as input `.Constraints.Bottom, width` and fails to handle the `, width` part of the string.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2022-03-16:
@wwmayer I think this is a duplicate of #5983 which is said to be solved by #3062

karlp, 2022-11-03:
This is still an issue in 0.20.1, FreeCAD 0.20.1, Libs: 0.20.1R29410 (Git)
Adding the exact error message to make finding this easier: ```ExpressionEngine: Invalid property specification```

I guess it's worth pointing out that the "bad" constraint names only matter if you have used parameter expressions for the value.

Constraint name contains "bad" chars, no parameters ==> works fine, name is truly "optional"
Constraint name contains "bad" chars, with parameters ==> works until you reload your file, then they're all gone. name very much not "optional" free text.

luzpaz, 2023-09-21:
See also 6593

realthunder's PR didn't get merged and therefore this issue and its relatives didn't get fixed