FreeCAD/FreeCAD #6599: [Bug] Loading a FCStd file containing a fillet produces Failed to load BRep file PartShapeXX.brp error

mfraser, 2022-03-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=67221

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 21.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28133 (Git)
Build type: Unknown
Branch: master
Hash: c0ceecbcf6ff396bca1307290caf48a51c6e8e22
Python 3.9.7, Qt 5.15.2, Coin 4.0.0, OCC 7.5.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * sheetmetal 0.2.49
  * lattice2 1.0.0
  * fcgear 1.0.0
  * Curves 0.3.0
  * Assembly4 0.11.8
  * fasteners
  * A2plus 0.4.54b
  * VendorParts
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

With the attached file, I get the error "Failed to load BRep file PartShape4.brp", if I remove the fillet, save the file and reload no error.

[Fillet test.tar.gz](https://github.com/FreeCAD/FreeCAD/files/8287452/Fillet.test.tar.gz)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mfraser, 2022-03-17:
Seems to also cause issues with chamfers, thickness & lofts.

GeneGH, 2022-03-17:
May be nothing wrong with the commit noted by OpenBrain. 

Rather, it seems that something earlier this year has been creating zero-size brep files in the FCStd file. Every one of my FCStd files created in the past few weeks has a zero-size brep file inside. I did not find any such files created in January or earlier.

donovaly, 2022-03-18:
Fixed with: https://github.com/FreeCAD/FreeCAD/commit/89e44550

closed 2022-03-18 by donovaly