FreeCAD/FreeCAD #6625: [Bug] [crash] Crash when creating a fillet

donovaly, 2022-03-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=581536#p581536

### Version

0.19 (Release)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +148 (Git)
Build type: Release
Branch: Branch_0.19.4
Hash: 476ecf091941bead59b14e44afa6064d5a66afa3
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: German/Germany (de_DE)
```

### Issue description

Even with @chennes fillet patches and OCC 7.5.3 FreeCAD crashes on this file: https://forum.freecadweb.org/download/file.php?id=183524
See here for the edge that causes the crash: https://forum.freecadweb.org/viewtopic.php?p=581536#p581536

### Anything else?

The crash is in FC 0.19 and latest 0.20 master

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2022-03-31:
Does this still crash with 7.6.1?

CharlemagneLasse, 2022-04-08:
@chennes, I can at least create such a crash with my example from https://forum.freecadweb.org/viewtopic.php?f=8&t=67870 with OCC 7.6.0. The problem about this crash was reported to @realthunder in https://github.com/realthunder/FreeCAD/issues/309

How can I get a build of realthunder's FreeCAD linkstage fork with OCC 7.6.1? I would like to test it

luzpaz, 2022-04-09:
@CharlemagneLasse you can use the [realthunder snap edge](https://snapcraft.io/freecad-realthunder) but there is an instability issue in the latest snap https://github.com/realthunder/FreeCAD/issues/307

CharlemagneLasse, 2022-04-09:
Thanks for the info. But yes, I have the same problem. So I've setup DRAWEXE from OCCT and created some tests as described in https://forum.freecadweb.org/viewtopic.php?p=587437#p587437
Unfortunately, it is not the same problem as I have seen in https://github.com/realthunder/FreeCAD/issues/309
But hey, when I am already here, why shouldn't I just the test for this specific example from ktostam. First step was to figure out when it crashes.

Crashed:

```
OS: Debian GNU/Linux 11 (bullseye) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28193 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 9c16923)
Hash: 9c1692336740c3bc7fa0c4d430ad5ceea223384f
Python 3.9.10, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: C/Default (C)
```

crashes:

```
OS: Debian GNU/Linux 11 (bullseye) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

So unfortunately, I don't have any never version to test it again. So let us directly switch to DRAWEXE to check if it can reproduce the problem or not

```
# get most recent OCCT
$ cd /tmp/
$ git clone https://git.dev.opencascade.org/repos/occt.git OCCT

# get build dependencies based on opencascade package in the Debian package source (deb-src) repository. will most likely work differently on whatever Distro you are using
$ sudo apt build-dep opencascade

# build everything (I only have 4 cores - you might change it if you are not working on a system with a 10 year old CPU)
$ cd OCCT
$ mkdir build
$ cd build
$ cmake .. -DCMAKE_BUILD_TYPE=Debug -DCMAKE_CXX_COMPILER_LAUNCHER=ccache -DCMAKE_VERBOSE_MAKEFILE=true  && make -j4

$ cd /tmp/
# figure out how the brp of the last Pad (name "Pad") is called:
$ unzip -p tutorial_excercise_1.FCStd Document.xml|xmllint --xpath 'string(//Document/ObjectData/Object/Properties/Property[@name="Label"]/String[@value="Pad"]/../../Property[@name="Shape"]/Part/@file)' /dev/stdin
PartShape3.brp

# extract the PartShape3.brp for later usage
$ unzip tutorial_excercise_1.FCStd PartShape3.brp

# start DRAWEXE shell
$ cd /tmp/OCCT/
$ cat > test.script << "EOF"
# load the brp, get edges and then try to add fillet on the problematic edge Edge10
pload ALL
restore /tmp/PartShape3.brp b
explode b E
blend b b 1 b_10
EOF
$ LD_LIBRARY_PATH=/tmp/OCCT/build/lin64/gcc/libd/ CASROOT=/tmp/OCCT ./build/lin64/gcc/bind/DRAWEXE
```
(If somebody wants to have the brp directly: [PartShape3.brp.gz](https://github.com/FreeCAD/FreeCAD/files/8457776/PartShape3.brp.gz) )

[... 66 lines cut ...]
$ cd OCCT
$ cat > test.script << "EOF"
# load the brp, get edges and then try to add fillet on the problematic edge Edge10
pload ALL
restore /tmp/PartShape3.brp b
explode b E
blend b b 1 b_10
exit
EOF

$ cat > testme.sh << "EOF"
#! /bin/sh
rm -rf build
mkdir -p build
cd build

(cmake .. -DCMAKE_BUILD_TYPE=Debug -DCMAKE_CXX_COMPILER_LAUNCHER=ccache -DCMAKE_VERBOSE_MAKEFILE=true  && make -j4) || exit 125
LD_LIBRARY_PATH=/tmp/OCCT/build/lin64/gcc/libd/ CASROOT=/tmp/OCCT valgrind -q --error-exitcode=23 --track-origins=yes --leak-check=no /tmp/OCCT/build/lin64/gcc/bind/DRAWEXE -f ../test.script || exit 1

EOF
$ chmod +x testme.sh

$ git bisect start V7_6_0 V7_3_0
$ git bisect run /tmp/OCCT/testme.sh
```

CharlemagneLasse, 2022-04-09:
There is also another similar crash in https://forum.freecadweb.org/viewtopic.php?p=584239#p584239
The brp for it can be found in [Body.Shape.brp.gz](https://github.com/FreeCAD/FreeCAD/files/8457882/Body.Shape.brp.gz)

How to reproduce the crash:

```
# get most recent OCCT
$ cd /tmp/
$ git clone https://git.dev.opencascade.org/repos/occt.git OCCT

# get build dependencies based on opencascade package in the Debian package source (deb-src) repository. will most likely work differently on whatever Distro you are using
$ sudo apt build-dep opencascade

# build everything (I only have 4 cores - you might change it if you are not working on a system with a 10 year old CPU)
$ cd OCCT
$ mkdir build
$ cd build
$ cmake .. -DCMAKE_BUILD_TYPE=Debug -DCMAKE_CXX_COMPILER_LAUNCHER=ccache -DCMAKE_VERBOSE_MAKEFILE=true  && make -j4

# start DRAWEXE shell
$ cd /tmp/OCCT/
$ cat > test.script << "EOF"
# load the brp, get edges and then try to add fillet on the problematic edges
pload ALL
restore /tmp/Body.Shape.brp b
explode b E
blend b2 b 1 b_10 1 b_18 1 b_19 1 b_20 1 b_21 1 b_17
EOF

# usually I would have done it like that. But it is then not 100% reproducible due to various system effects.
# So I just use gdb in the actual command to improve my live and make it 100% reproducible
#$ LD_LIBRARY_PATH=/tmp/OCCT/build/lin64/gcc/libd/ CASROOT=/tmp/OCCT ./build/lin64/gcc/bind/DRAWEXE -f test.script

$ LD_LIBRARY_PATH=/tmp/OCCT/build/lin64/gcc/libd/ CASROOT=/tmp/OCCT gdb -q --ex run --args ./build/lin64/gcc/bind/DRAWEXE -f test.script
```
* @blobfish V7_6_0-12-gac645430b3: [crash2-V7_6_0-12-gac645430b3.txt](https://github.com/FreeCAD/FreeCAD/files/8457892/crash2-V7_5430b3.txt)
6_0-12-gac64
* 7.6.0: [crash2-V7_6_0.txt](https://github.com/FreeCAD/FreeCAD/files/8457898/crash2-V7_6_0.txt)
* 7_6_0-122-g2491eec38b (master): [crash2-V7_6_0-122-g2491eec38b.txt](https://github.com/FreeCAD/FreeCAD/files/8457896/crash2-V7_6_0-122-g2491eec38b.txt)
* 7.1.0: [crash2-V7_1_0.txt](https://github.com/FreeCAD/FreeCAD/files/8457901/crash2-V7_1_0.txt)

luzpaz, 2022-04-09:
Just a side note: This is such impressive and extensive testing. #nextlevel

chennes, 2022-04-10:
So we can mitigate the crash by checking TF before accessing it: it won't make the fillet work, but it will resolve the hard crash. Do you want to register as an OCCT developer and submit that patch via their PR system? Just attaching it to a bug report won't work, they won't use the code there.

CharlemagneLasse, 2022-04-10:
@chennes, I am not a CG developer and have no idea how these algorithms are supposed to work. So the two workarounds which I've posted at the end of the comments is the best which I can make without studying the algorithms in detail.

I've tried to register with their bug tracker (which created a single-sign on account) but I am unable to use this account on the bugtracker. Instead of going to the single sign on (like the register button), it opens another login page with username + password which doesn't accept the single sign on credentials. Not sure what it broken at @Open-Cascade-SAS
I've also used the contact form but didn't receive any answer yet. Maybe @smoskvin, @alexander-malyshev, @gkv311 or @ssafarov can assist here. At least it is should not be in their interest to have a crashing product just because people cannot report the two problems to them.

People like @coldtobi and @kkremitzki from Debian or @blobfish might still handle that in their downstream projects until the upstreaming problems + reporting with OCCT / @Open-Cascade-SAS are solved.

And for the CLA: I have no problem when they relicense and copyright code which I sent to them for this OCCT. But their CLA requires a mail address and telephone. Things which I currently don't have (slightly complicated situation which is hard to explain). I can still try it later and in worst case ask some fellow entities whether I can use their maker/hackerspace as address. But the changes here are not copyrightable anyway because they have no added value (besides fixing obvious errors) and thus wouldn't create a new body of work that reaches the threshold of originality.

CharlemagneLasse, 2022-04-13:
Bug was forwarded to:

* https://tracker.dev.opencascade.org/view.php?id=32930 (crash + workaround)
* https://tracker.dev.opencascade.org/view.php?id=32931 (crash + workaround)
* https://tracker.dev.opencascade.org/view.php?id=32932 (failed fillet)
* https://tracker.dev.opencascade.org/view.php?id=32933 (failed fillet)

donovaly, 2022-05-26:
For FreeCAD 0.20 we cannot do anything -> removing the milestone

luzpaz, 2022-07-29:
Good news @CharlemagneLasse !

> [tracker.dev.opencascade.org/view.php?id=32930](https://tracker.dev.opencascade.org/view.php?id=32930) (crash + workaround)
> [tracker.dev.opencascade.org/view.php?id=32931](https://tracker.dev.opencascade.org/view.php?id=32931) (crash + workaround)

Have both been fixed upstream and are assigned to OCC 7.6.3 ([Released](https://github.com/Open-Cascade-SAS/OCCT/releases/tag/V7_6_3) 2 days ago!)

But, no traction on:
> [tracker.dev.opencascade.org/view.php?id=32932](https://tracker.dev.opencascade.org/view.php?id=32932) (failed fillet)
> [tracker.dev.opencascade.org/view.php?id=32933](https://tracker.dev.opencascade.org/view.php?id=32933) (failed fillet)

They've been assigned but no visible progress

luzpaz, 2022-07-31:
Hi @CharlemagneLasse would you mind bumping your upstream tickets? They didn't get any response from the OCC team

> [tracker.dev.opencascade.org/view.php?id=32932](https://tracker.dev.opencascade.org/view.php?id=32932) (failed fillet)
> [tracker.dev.opencascade.org/view.php?id=32933](https://tracker.dev.opencascade.org/view.php?id=32933) (failed fillet)

donovaly, 2022-08-11:
> Have both been fixed upstream and are assigned to OCC 7.6.3 ([Released](https://github.com/Open-Cascade-SAS/OCCT/releases/tag/V7_6_3) 2 days ago!)

The Windows build of FreeCAD 0.20.1 comes with OCC 7.6.3. Thanks to @blobfish for his work!

adrianinsaval, 2023-03-22:
from my understanding the crash has been solved and the only remaining issues here are that the fillets don't work, is it worth keeping this open?

If yes, I suggest changing the title to reflect the current state

donovaly, 2023-03-22:
You are right -> closing it.

closed 2023-03-22 by donovaly