FreeCAD/FreeCAD #6633: zoom icons wrong in KDE

Giorgio-IT, 2022-03-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=67336

### Version

0.20 (Development)

### Full version info

```shell
OS: OpenMandriva Lx 4.3 (Dysprosium) Official (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.Unknown
Build type: RelWithDebInfo
Python version: 3.9.8
Qt version: 5.15.3
Coin version: 4.0.0a
OCC version: 7.6.0
Locale: Italian/Italy (it_IT)
```

### Subproject(s) affected?

Core

### Issue description

Two icons related to zoom are missing in Linux, it seems particularly in KDE environment.

This is how it appears in OpenMandriva LX and openSuse, both using KDE
[image]
This is how it should be (and is in Appimage in Linux Mint)
[image]

### Anything else?

This bug is present in previous versions too. At least 0.19 and 0.18.
This is the same [bug in KDE](https://bugs.kde.org/show_bug.cgi?id=450870) I opened with interesting answers.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-03-21:
Related: https://forum.freecadweb.org/viewtopic.php?f=8&t=65257

donovaly, 2022-03-21:
So this issue is something in KDE, meaning FreeCAD is not to blame?

luzpaz, 2022-03-21:
There is a workaround in FreeCAD. But yes, this an upstream issue. We should keep this open if there is an upstream ticket.

Pointedstick, 2022-03-23:
Yes, that's an example of the problem caused by https://bugs.kde.org/show_bug.cgi?id=435671. We'll fix that in Breeze Icons eventually.

Pointedstick, 2022-03-23:
So you can probably close this since I don't think there's anything to fix on your side.

luzpaz, 2022-03-23:
@Pointedstick are you a dev working on the KDE breeze related things?

Pointedstick, 2022-03-23:
Yep!

donovaly, 2022-03-23:
> So you can probably close this since I don't think there's anything to fix on your side.

Thank you. Closing.

closed 2022-03-23 by donovaly