FreeCAD/FreeCAD #6641: [Bug] [regression] polar pattern fail

donovaly, 2022-03-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=19&t=67161

### Version

0.20 (Development)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.28276 (Git)
Build type: Release
Branch: master
Hash: 56ced5cdd594d03c85dfa22c45bbabc0d96b2676
Python 3.8.6+, Qt 5.15.2, Coin 4.0.1, OCC 7.5.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

PartDesign

### Issue description

Polar patterns fail in FC 0.20. Bisecting lead to this commit as the cause: https://github.com/FreeCAD/FreeCAD/commit/03da9d3501

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2022-03-21:
I reverted https://github.com/FreeCAD/FreeCAD/commit/03da9d3501 but the bug persists

donovaly, 2022-03-21:
I reverted https://github.com/FreeCAD/FreeCAD/commit/03da9d3501 but it is indeed the commit to blame

AjinkyaDahale, 2022-03-24:
@donovaly I tested your revert, and it seems to reset the items in pattern to 3 again.

donovaly, 2022-03-24:
> and it seems to reset the items in pattern to 3 again.

Yes, I noticed this too. Just the reversion is too much here. However the reversion created at least a pattern.

So this is a big mess. I spent already few hours without making progress :-(

AjinkyaDahale, 2022-03-24:
> Yes, I noticed this too. Just the reversion is too much here. However the reversion created at least a pattern.

It's odd. I don't even see changes in `Restore`.

AjinkyaDahale, 2022-03-27:
Note to self and anyone else attempting this: I predict there will also be a problem for certain cases where the same "tool" adds AND removes something from the object. This is not yet observed because this but itself is not completely resolved, but when it is, take a look at
https://github.com/FreeCAD/FreeCAD/blob/7676216a7a9b501c2cfad81fa4ce3c5585898479/src/Mod/PartDesign/App/FeatureDressUp.cpp#L268
and
https://github.com/FreeCAD/FreeCAD/blob/7676216a7a9b501c2cfad81fa4ce3c5585898479/src/Mod/PartDesign/App/FeatureTransformed.cpp#L297

Notice that in `FeatureDressUp.cpp#L268` both `addShape` and `subShape` can be non-null. These correspond to `fuseShape` and `cutShape` in `FeatureTransformed.cpp#L297`, where only one or the other is applied. This probably did not happen prior to 03da9d3501 since there `cutShape.isNull()` is also checked in the `if (!fuseShape.isNull())` branch.

A possible test case would be one similar to the one in https://forum.freecadweb.org/viewtopic.php?f=19&t=67161&start=10#p582840 where a single fillet replaces both the fillets.

adrianinsaval, 2022-03-28:
can fillets have an addsubshape with mixed possitive and negative volume shapes? Perhaps this is a way to solve this, it can be always treated as substractive and if the additive parts have negative volume they would be added instead.

0penBrain, 2022-04-06:
> @0penBrain sorry for the late reply

@adrianinsaval

adrianinsaval, 2022-04-06:
I was referring to boolean operations as BO. With fillet instability do you mean the topo naming problem or something else? IMO waiting for topo naming merge is not a good idea because as we've seen in the past it can be years before it gets merged.

AjinkyaDahale, 2022-04-07:
> With fillet instability do you mean the topo naming problem or something else?

This is not related to the topo-naming issue. I was referring to https://forum.freecadweb.org/viewtopic.php?f=19&t=67161&start=10#p582840 and following posts on the thread. You can try the file from https://forum.freecadweb.org/viewtopic.php?f=19&t=67161&start=30#p583583 and change the number of repetitions to see what I meant by "instability".

donovaly, 2022-04-12:
Update:
- the Polar pattern issue was resolved by: https://github.com/wwmayer/FreeCAD/commit/6662a0afdb6
- the bug is that Pipe and Loft cannot be transformed anymore: https://forum.freecadweb.org/viewtopic.php?p=587977#p587977

AjinkyaDahale, 2022-04-12:
@donovaly this can probably be closed.

donovaly, 2022-04-13:
Cllosing. Fixed with: https://github.com/FreeCAD/FreeCAD/commit/5ab7d17f679

closed 2022-04-13 by donovaly