FreeCAD/FreeCAD #6644: [Bug] FIX freecad-daily Ubuntu 22.04 requires libvtk9, but libvtk9.1 provided

allanlaal, 2022-03-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

what? no!

### Version

0.20 (Development)

### Full version info

```shell
deb http://ppa.launchpad.net/freecad-maintainers/freecad-daily/ubuntu/ jammy main
```

### Issue description

cannot `apt install` in 22.04 jammy, since it now provides only libvtk9.1, but freecad-daily requires 9.0

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2022-03-22:
We have good reasons to only accept bug reports with prior discussions. Your issue can be a packaging issue and then the question is where the bug is.
Therefore please open a thread in the help forum. When others can reproduce (we have many Ubuntu users around), then this issue can e reopened.

luzpaz, 2022-03-24:
Using the unofficial snap `snap info freecad-ppd` (since vejmarie hasn't been hands on with maintaining the official snaps, ppd stepped in and made the unofficial snap)

luzpaz, 2022-03-24:
> isn't `http://ppa.launchpad.net/freecad-maintainers/freecad-daily/` that PPA ?
> 

Of course, my bad. I missed that point. The PPA is maintained by us. @kkremitzki can you weigh in on OP's issue?

allanlaal, 2022-03-24:
perhaps a solution would be to include libvtk9 (9.0) in the ppa?

luzpaz, 2022-03-24:
@allanlaal vtk is can be problematic to update.

donovaly, 2022-03-24:
Guys, please follow our rules. Open a forum thread and discuss there. Once it is clear that it is a FreeCAD bug, reopen this issue.
In every case the discussion should be in the forum's Help subforum so that more users can contribute.

closed 2022-03-24 by donovaly

luzpaz, 2022-03-24:
Continued in https://forum.freecadweb.org/viewtopic.php?f=42&t=67432

donovaly, 2022-03-24:
> Continued in https://forum.freecadweb.org/viewtopic.php?f=42&t=67432

Thanks.