FreeCAD/FreeCAD #6653: [Bug] [TD] SVG import does not work with plain SVG

donovaly, 2022-03-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=35&t=67413

### Version

0.20 (Development)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.28343 (Git)
Build type: Release
Branch: master
Hash: b0b243bee79cd5924210e2c8241cde2d30b51ab1
Python 3.8.6+, Qt 5.15.2, Coin 4.0.1, OCC 7.5.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Techdraw

### Issue description

- take this plain SVG of a box (created using Inkscape 1.1 and then saved as "Plain SVG"):

and import it as SVG

result:
- you get an "error" about a missing DPI setting. However the settings are the SVG (as in the SVG specs it contains the mm settings).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2022-03-24:
Why is this tagged for TechDraw?  "Insert Svg Symbol" imports the file without complaints.  

The error messages only appear on import as geometry or import to Drawing wb.

donovaly, 2022-03-24:
Thanks, I was not aware this is a Drawing WB issue.

yorikvanhavre, 2023-05-08:
Will look at it again

PR #9533 (2023-05-13): Draft: Skip SVG import DPI dialog if a unit is specified in the file
This fixes #6653

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR
commit c4a6b3a0 (2023-05-13): Draft: Skip SVG import DPI dialog if a unit is specified in the file - fixes #6653 (#9533)
closed 2023-05-13 by Roy-043