FreeCAD/FreeCAD #6655: [Bug] [regression] App::Part Shapes are infinite

WandererFan, 2022-03-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://www.forum.freecadweb.org/viewtopic.php?f=20&t=67202

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.
Build type: Release
Branch: unknown
Hash: c360c6e5f5a8cee3bd85c01cf97128a0b9e0e169
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, OCC 7.5.2
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.54b
  * Curves 0.3.0
[/code]
```

### Subproject(s) affected?

Techdraw

### Issue description

In v020, App::Part that contains one or more LCS has an infinite shape.  The infinite shape can not be projected and therefore can not be displayed in TechDraw. 

In v019, the same App::Part has a finite shape which can be projected.

We need to either restore the v019 behaviour or develop a new function to disassemble App::Part into its components, skip the LCS and reassemble the shape for projection.

### Anything else?

[Technical forum discussion](https://www.forum.freecadweb.org/viewtopic.php?f=10&t=67320)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2022-04-03:
After applying the changes of 484ff4fafb607d to a v0.19 build then the exact same problems occur. So, this means the behaviour of Part or PD hasn't changed in this context.

WandererFan, 2022-04-14:
Not a regression.  Shape was infinite in 0.19 but the infinite plane was not visible.

adrianinsaval, 2022-04-15:
didn't you say that techdraw already handles infinite shapes now? can this still be considered a bug?

WandererFan, 2022-04-16:
I think it is a bug.  The LCS is an artificial construct.  It shouldn't be included in the shape.

WandererFan, 2022-05-26:
> Can we do something here for FreeCAD 0.20?

We implemented a work around in TechDraw a while ago.  It now searches the
input shapes and removes anything that is infinite.  So as far as TD is
concerned, this is a non-issue.  Nobody else seems to find infinite shapes
an issue, so I guess we can close the issue.

adrianinsaval, 2022-05-26:
I agree that conceptually the datums shouldn't be in the Part shape so perhaps the issue can be kept open, but I would remove it from the 0.20 milestone

donovaly, 2022-05-26:
> I agree that conceptually the datums shouldn't be in the Part shape

But this is contrary to

> Nobody else seems to find infinite shapes an issue

Let's do it this way. let's close this for now and if there is later a consensus that infinite shapes are an issue, this issue can be reopened.

closed 2022-05-26 by donovaly