FreeCAD/FreeCAD #6663: [Bug] Decimal separator not chosen by locale system settings anymore (dot instead of comma)

fiechr, 2022-03-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

Can't use forums: Get "General Error Module not accessible", when trying to log in

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28366 (Git)
Build type: Release
Branch: master
Hash: c4576ecfeb6cfa056381b05e0f14e1b244955020
Python 3.10.2, Qt 5.15.3, Coin 4.0.1, OCC 7.5.3
Locale: C/Default (C)
Installed mods: 
  * Render 2022.1.0
[/code]
```

### Issue description

I've used FreeCAD on Linux (Manjaro/Arch) for multiple years and up until now, it did exactly what I was supposed to, by using English as UI language and still used my configured local number format (German) for displaying numbers.

As you may already know, Germany uses a "," [comma] instead of a "." [dot] as decimal separator.

Unfortunately, this seems to be suddenly broken, since I now **have** to use the dot, since my locale settings and KDE/Plasma settings seem to be ignored for now.

I only found an option to replace the numpad comma with a dot, but that's even worse. We don't have English keyboards in Germany and if I press on a comma, I expect a comma to show up on screen. So I really hope this is just a bug and not how it's supposed to be.

These are my locale settings (which worked fine before):
```
LANG=en_US.UTF-8
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=de_DE.UTF-8
LC_TIME=de_DE.UTF-8
LC_COLLATE=C
LC_MONETARY=de_DE.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=de_DE.UTF-8
LC_NAME=de_DE.UTF-8
LC_ADDRESS=de_DE.UTF-8
LC_TELEPHONE=de_DE.UTF-8
LC_MEASUREMENT=de_DE.UTF-8
LC_IDENTIFICATION=de_DE.UTF-8
LC_ALL=
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-03-26:
Please follow the developments of #6640  
Are you saying the forum is down? I don't seem to have an issue accessing them

fiechr, 2022-03-26:
> Are you saying the forum is down? I don't seem to have an issue accessing them

No not down, but as soon as I entered my username and password I got the above mentioned error message. I've tried it multiple times. But the login seems to work now again, but the forum looks different (without custom theme) when I click in the link in the menu of https://www.freecadweb.org/.

> Please follow the developments of #6640

Thank you for the link! I'm sorry, but I missed this one. I was searching for "locale" and "decimal separator" and it didn't showed up.

donovaly, 2022-03-27:
We fixed the known issues now. Please check again with build >= Version: 0.20.28450 (Git)

If the problem persists, please reopen.

fiechr, 2022-03-27:
I've rebuilt the version shown below, but for me all decimal separators are still "." (dot) instead of "," (comma).
Commas are simply ignored, so 23,21 mm becomes 2321 mm.

[image]

```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28452 (Git)
Build type: Release
Branch: master
Hash: b7b2d73b023a3825c9c126969e35a912b9467ddd
Python 3.10.2, Qt 5.15.3, Coin 4.0.1, OCC 7.5.3
Locale: C/Default (C)
Installed mods: 
  * Render 2022.1.0

```

fiechr, 2022-03-27:
For the record, this is how it looked before, when it worked:
(Its an older package, I built on date 2022/03/08)

[image]

```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.27976 (Git)
Build type: Release
Branch: master
Hash: 84f23401709f2efb95bfb59680b8746b945136be
Python 3.10.2, Qt 5.15.2, Coin 4.0.1, OCC 7.5.3
Locale: English/United States (en_US)
Installed mods: 
  * Render 2022.1.0
```

donovaly, 2022-03-27:
> I've rebuilt the version shown below, but for me all decimal separators are still "." (dot) instead of "," (comma). Commas are simply ignored, so 23,21 mm becomes 2321 mm.

But your screenshots look correct. You use English menus in FreeCAD, right? Thus the decimal separator is the dot, not the comma. Change it to German menus and you can use the comma.

That even with English menus one had to use the comma on a German OS was the initial bug we fixed.

p.s. please create a forum account. I checked that it works, just assure that you use this link:
https://forum.freecadweb.org/

fiechr, 2022-03-27:
Yes, but that's the whole point. I don't want the German language, I want English. Only the number format is supposed to be German. Like it was before (see last screenshot).

Thats what locales are there for, they allow separation of language and different other formats, like numbers or currency, paper sizes and stuff like that.

I'm using Linux, Windows, Android and iPadOS with the English language, while numbers etc. are still formatted/used according to the German locale. This works perfectly fine. Setting the number format according to the selected language is just bad practice.

donovaly, 2022-03-27:
>I don't want the German language, I want English. Only the number format is supposed to be German. Like it was before (see last screenshot).

As I wrote, users often complained about this. Everything is in English, but the numbers are not. This could not be changed and people could e.g. not input values from their Excel sheets.

What you want would be to add a new combobox to select the number format independently from the menu language. This feature did no exist in FreeCAD. For example to have German menus but using the dot as decimal separator.

Now you can have German menus but using the dot as decimal separator, only having English menus but having the comma does not work.

However, I am a bit baffled that you like to work with this since it is error-prone as hell. People see English menus and don't expect to use German decimal settings. At my work this issue cost me in summation days  to fix, especially with spreadsheets.

fiechr, 2022-03-27:
Thank you for time and work regarding this issue and FreeCAD as a whole, I really appreciate it!

I don't think language settings inside FreeCAD are needed at all, it just needs to follow the settings of the OS. All operating systems I know and use, allow to set language and things like date format or number formats to different values.
It makes sense, because just because someone is living or residing in a certain country, doesn't mean that she or he speaks or wants to use the language of that country. But that's not the only reason.

I like to keep everything in English, because often translations are bad or incomplete, also German words often are longer and need to be abbreviated, up until to the point of incomprehensibility (that has nothing to do with FreeCAD, it's just my experience).
It's also a lot easier for getting help online or contribute to open source software, since the names of the menu entries, functions or error messages are already in the language that the developers (most of the time) use, which is English.

The big ones support different language and date/number formats etc. for years, some even for decades. And I can confirm that I have very rarely any issues with using English with German dates and numbers. Even LibreOffice works completely fine with English UI and German formats, including the spell checker. I can easily switch between different languages per document, without changing the language of the UI I'm used to.

Of course there are **always** people who don't get it and maybe in a corporate or business setup, it may make more sense to limit the language to one for everyone (at least in a certain branch). But that's only one use case of many.

In my opinion, it makes no sense to restrict or force something that doesn't need to be restricted, especially when it already worked for years and the facilities are already proven and in place.

But, for whatever reason, if an additional setting inside FreeCAD is preferred, I would really really appreciate the possibility to use a different number format, without being forced to change the UI language for that as well.

donovaly, 2022-03-27:
> I don't think language settings inside FreeCAD are needed at all, it just needs to follow the settings of the OS.

But this did not work. For example at work we have German Win 10. My Indian colleagues set FC's menus to English but still had to use the comma as decimal separator. There was no way to change this. Windows' settings does not allow you to change to English numbers. Here at home, where I have admin permissions, I cannot even find this setting. I can only change the whole number system , for example the date format. But this will break the calendar apps.
So all in all I find it very helpful to have a whole program in one language.

However, as also recently pointed out by @wwmayer , we should offer an option to get the previous behavior back that allowed you to use English menus but German numbers.

Reopening and inviting @OpenBrain

wwmayer, 2022-03-28:
> I've rebuilt the version shown below, but for me all decimal separators are still "." (dot) instead of "," (comma).
Commas are simply ignored, so 23,21 mm becomes 2321 mm.

In the preferences there is the option "Use selected language number format". If it's off then the C locale is used and if it's on then the locale is used that is bound to the language. Since you use English as language the locale is set to something else than German locale. So, effectively the option to use English language but German locale is not supported any more and IMO indeed a regression. I already wrote in the forum that the locale settings of the OS should be used by default, i.e. the behaviour how it was before this feature was introduced.

> I don't think language settings inside FreeCAD are needed at all, it just needs to follow the settings of the OS.

Well, apparently there are users who wanted this feature but this doesn't mean to remove the old behaviour.

I will work on the user settings to bring back the old behaviour...

closed 2022-03-28 by wwmayer

fiechr, 2022-03-28:
I've just rebuilt and can confirm, that it works again: English language and comma separated decimal values.

Thank you very much!

commit bba7719c (2022-03-28): Gui: fixes #6663: [Bug] Decimal separator not chosen by locale system settings anymore