FreeCAD/FreeCAD #6672: [Bug] PartDesign SubShapeBinder: crash and other issues when manually editing Support

Roy-043, 2022-03-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=67401&start=10#p582669

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Windows 8.1 (6.3)
Word size of FreeCAD: 64-bit
Version: 0.20.28193 (Git)
Build type: Release
Python 3.8.12, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: C/Default (C)
Installed mods: 
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

There are some issues when you manually edit the Support property of a SubShapeBinder.

To make the Support property editable:

1. Right-click in the Property editor.
2. Select "Show All".
3. Right-click the Support property.
4. De-select "ReadOnly".
5. Click in another field.
6. Click in the Support field.
7. Press the "..." button.
8. The Link dialog opens.

A.

1. Open the attached file.
2. Make the Support property editable as already explained.
3. Click the "..." button in the Support field.
4. In the Link dialog press the "Clear" button.
5. Select the highest Cylinder entry in the dialog list (not under Part).
6. Press "OK".
7. Result: FC crashes.

B.

1. Open the attached file.
2. Drag and drop the Cube onto the SubShapeBinder.
3. Make the Support property editable.
4. Click the "..." button in the Support field.
5. In the Link dialog do nothing just press the "OK" button.
6. Recompute.
7. Result: The Support field now only contains a reference to the face of the cylinder, but in the 3D view the Cube is still shown as part of the SubShapeBinder.

C.

1. Open the attached file.
2. Delete the existing SubShapeBinder.
3. Make the Part visible.
4. Select the top face of the Cylinder.
5. Click the [PartDesign_SubShapeBinder](https://wiki.freecadweb.org/PartDesign_SubShapeBinder) button.
6. Result: Support is "Part.Cylinder.Face3".
7. Again delete the SubShapeBinder.
8. Make sure nothing is selected.
9. Click the [PartDesign_SubShapeBinder](https://wiki.freecadweb.org/PartDesign_SubShapeBinder) button again.
10. Make the Support property editable.
11. Click the "...' button in the Source field.
12. In the 3D view select the top face of the Cylinder.
13. Result: Support is "Cylinder.Face3". Without the nesting in the Part. Note that when selecting the face the statusbar does show the nesting in Part.

[test-ssb.zip](https://github.com/FreeCAD/FreeCAD/files/8361305/test-ssb.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2022-04-03:
Case A is most like fixed already, as I cannot reproduce the crash.

B is property item display problem. The `Support` contains two entries, but only one is displayed. It's fixed in one of the commit above.

C is a problem regarding the `Sync sub object` option in the property editor dialog. With this option checked, the sub-object path is captured from 3D view selection. The commit above auto enables this option with `PropertyXLinkSubList` which is the type of `SubShapeBinder.Support`.

wwmayer, 2022-04-10:
I can confirm that B+C is fixed by the PR #6710 but the crash isn't. The crash is fixed with 1ff4ad085c2

PR #6710 (2022-04-10): App/Gui: fix property link editor
Fixes #6672
closed 2022-04-10 by wwmayer