FreeCAD/FreeCAD #6678: [Bug] <Drag & drop fail... Access violation>

bntarc, 2022-03-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=20&t=67331

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.28193 (Git)
Build type: Release
Python 3.8.12, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.55a
  * Assembly3 0.11.0
  * BOLTSFC
  * CurvedShapes 1.0.1
  * dxf-library
  * frame
  * parts_library
  * Reporting
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Issue description

I am using the workbench Assembly 3.
When I try to Drag & drop a part to an assembly (the assembly and part are 2 different file) I have the following message:
AS3 Drag & drop fail... Access violation

However when I import a part to an assembly (the assembly and part are in the same file) is is working.

0.20-27809
11:58:02 Traceback (most recent call last):
File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\Assembly3\.\freecad\asm3\assembly.py", line 469, in dropObjectEx
me.setLink({-1:obj})
<class 'Base.FreeCADError'>: Access violation

0.20-27422
11:53:44 Unknown command 'Part_SectionCut'
11:55:04 Illegal storage access...
11:55:04 Traceback (most recent call last):
File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\Assembly3\.\freecad\asm3\assembly.py", line 469, in dropObjectEx
me.setLink({-1:obj})
<class 'Base.FreeCADError'>: Illegal storage access! Please save your work under a new file name and restart the application!

0.20-28193
12:00:04 Recent macros : shortcut 1 disabled because conflicting with Std_ViewFront
12:00:04 Recent macros : shortcut 2 disabled because conflicting with Std_ViewTop
12:00:51 Traceback (most recent call last):
File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\Assembly3\.\freecad\asm3\assembly.py", line 469, in dropObjectEx
me.setLink({-1:obj})
<class 'OSError'>: Access violation

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-03-29:
Wait, is this strictly an Assembly3 (external) workbench issue?

donovaly, 2022-03-30:
This is an Assembly 3 workbench issue. Please report it here:  https://github.com/realthunder/FreeCAD_assembly3

closed 2022-03-30 by donovaly

bambukouk, 2022-03-30:
> Wait, is this strictly an Assembly3 (external) workbench issue?

It only affects master branch 20
works fine in master 19 and Link branch

luzpaz, 2022-03-30:
Are you sure @donovaly it's strictly A3 and doesn't relate to FreeCAD master?

bambukouk, 2022-03-30:
latest:
https://forum.freecadweb.org/viewtopic.php?f=20&t=67331#p584316

donovaly, 2022-03-30:
> Are you sure @donovaly it's strictly A3 and doesn't relate to FreeCAD master?

I cannot reproduce a crash. can you? if so, what do I have to do to reproduce in current master?

bambukouk, 2022-03-30:
I have tested and failed (at the time) using 28193, but things are moving fast...
and it would appear (as per link quoted in my last post) that with the latest being 28445 
.`..seems to be fixed again in the meantime....` as per @wwmayer 
so the issue might be history?

wwmayer, 2022-03-31:
The crash was fixed with d1b6bb78d124181b6

bntarc, 2022-10-11:
I had this bug on Assembly3. I don't know if it there is the same bug on
other workbench.