FreeCAD/FreeCAD #6692: [Feature Request] Enable KeyboardInterrupt (ie Ctrl-C) to work in the Python console to break out of infinite loops

goatchurchprime, 2022-03-30:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=22&t=67587&p=584414

### Subproject(s) affected?

Core

### Idea description

The Python console of FreeCAD treats Ctrl-C as a copy selection instead of a KeyboardInterrupt.  

This makes it impossible to break out of an infinite loop or long running process as you might have to do when you are hacking Python macro code. 

(There is some process in the system which detects that the main GUI thread has frozen and pops up a dialog box with the option to kill the entire FreeCAD process or continue to wait.  It would be nicer if this dialog box also had the option to send a signal.CTRL_C_EVENT into the Python code.)

It seems that the Python macro code is running directly in the single GUI thread, which is why everything completely freezes.  A work-around [mentioned in the forum](https://forum.freecadweb.org/viewtopic.php?t=12678#p101749) is to call back out to some function in the FreeCADGui class to poll a Cancel button in a popup dialog.  But this is no use if your code is stuck in a loop that doesn't include this function.

 

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MicobyteMichael, 2022-04-07:
I believe I've addressed this issue in PR #6720 by moving execution of Python code to a background thread and adding a Ctrl+C listener to the GUI. Please see the PR for details.

goatchurchprime, 2022-04-12:
Thinking a bit more about this, it might be a good idea to implement this with a pop-up window (instead of the Ctrl-C) similar to the "FreeCAD is not responding" one, but to include the option to interrupt/cancel the running Python macro instead of Force quit the whole application.

Such a popup would automatically appear when a macro runs for more than five seconds and it would show a timer for how long it had been running.

(As a side-note, in one CADCAM system I once worked on we had to remove the progress timer because too many users used it to measure the calculation speed in the next version and complained if it ran one second longer: "Why am I paying this much money for an upgrade that's slower?".  Unlike proprietary software customers, open source users can handle the truth because they know how to send back useful information to the developers rather than just refuse to pay the invoice!)

This Macro-In-Progress pop-up would have space for a percentage progress bar, which would make it easy for macro developers to learn than they need to make a call-back to set the progress value every so often so that the user gets an idea for how long it's got left to run.

luzpaz, 2023-09-09:
#6720 replaced with #10587

luzpaz, 2023-09-09:
CC @FlachyJoe

luzpaz, 2023-09-12:
bumping so we don't lose sight of this ticket

FlachyJoe, 2023-09-12:
@luzpaz
#10587 is merged and fixes this.
#10619 is merged and fixes #10617

luzpaz, 2024-12-02:
@goatchurchprime can you verify this is fixed in 1.0.0 ?

goatchurchprime, 2025-01-12:
> @goatchurchprime can you verify this is fixed in 1.0.0 ?

Finally got round to checking it, now with it in nixos.  It surely works, whereas before it didn't.  

[image]

closed 2025-06-08 by luzpaz