FreeCAD/FreeCAD #6697: [Feature Request] Backups for each project stored in their own directory to avoid directory bloat

luzpaz, 2022-04-01:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

tbd

### Subproject(s) affected?

Core

### Idea description

This ticket is a continuation and a consequence to [#5451](https://github.com/FreeCAD/FreeCAD/pull/5451)
When increasing the amount of minimum backups, FreeCAD then autogenerates them all in the same place willy-nilly which causes unnecessary filesystem disorganization. 

This FR continues exploring the way to deal with this symptom. 
  

### Anything else?

Related to #6045

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2022-04-02:
I suggest a hidden .fcbak directory, for each directory, not per project.

For example you'll have this structure
/a/directory/path
|____file1.fcstd
|____file2.fcstd
|____.fcbak
         |____file1.fcstd1
         |____file1.fcstd2
         |____file2.fcstd1

luzpaz, 2022-04-02:
@adrianinsaval do you know if dotfiles 'hidden' on windows as well ?

adrianinsaval, 2022-04-02:
No, they need to be set to hidden

luzpaz, 2022-04-02:
@adrianinsaval can we do that when we create them programmatically ?

adrianinsaval, 2022-04-02:
Most likely yes if the filesystem supports it, AFAIK ext4 and co don't support that so if you create it on Linux FS and the copy to a windows fs it might not be hidden. But that isn't FreeCAD's responsibility.
We should also take into account if we actually want the backups hidden, as noobs won't immediately know they exist, perhaps it should be a regular directory called FCBak, and ideally the backup path should be configurable.

aslze, 2023-02-22:
I was about to post a similar issue (the ugly directory listings with pairs of *.FCStd and *.FCStd1) but not necessarily with the same proposed solution. I think, the proposed solution will also create clutter with hidden directories in each directory with FC files.

What others do:

* Blender stores backup "autorecovery" files in the Temp directory (per user). After all you don't need these files forever and the temp directory can be cleaned regularly.
* MS Word creates "autorecovery" files in the user's profile (AppData/Roaming/MS/Word). Not exactly backup files. Seem to be special recovery info.

I think the goal of these backups is to avoid data loss if FreeCad crashes or if you exit and forget to save, isn't it? (the latter can be solved with a warning dialog). So I don't want these files to lay around forever. Why not delete them when exiting FreeCAD cleanly and if the opened files were saved fine? This way they could be in the same directory, maybe hidden. Or they can go to a single special directory in the user's profile FreeCAD directory.

adrianinsaval, 2023-02-22:
No these are backups of the previous version of the file, autorecovery files are stored in a temp directory too.

FlachyJoe, 2023-02-22:
@aslze backup policy can be tweak to store many old version and so used as a time machine.
If you don't need the backup you can already uncheck _preference/General/Document/Maximum number of backup files…_

While the recovery files are available I can't see reason to store a backup EXCEPT to revert to a previous **saved** state. As this action is triggered by user, he expects to easily find these files. So I wouldn't hide them.

aslze, 2023-02-22:
Oops. My misunderstanding. Thanks.
I can't find the autorecovery files anywhere, though. But they do work.

I second keeping them in a separate directory to keep things clean. Either a subdirectory or a global backup directory.

MatthiasWM, 2023-05-07:
I can offer to implement this. My understanding is that I would add a global preferences setting for a path name relative to the current project directory, that may or may not contain the name part of the document filename. Then in App::BackupPolicy, I would make sure the target directory exists and then add the path component to the target name in applyStandard() and applyTimeStamp().

Is that correct? Would you like me to make a PR for that addition?

adrianinsaval, 2023-05-26:
> Is that correct? Would you like me to make a PR for that addition?

Sorry for the late response, of course yes, PRs are welcomed!

I suggest FCBak as default path, it would be nice to allow specifying an absolute path instead of one relative to the file (a user might like to have all backups in the same place).

MatthiasWM, 2023-05-29:
Thanks for the feedback. I am busy travelling ATM, but I will try to get a rough preliminary PR out soon.

rebeltaz, 2023-06-12:
> I suggest a hidden .fcbak directory, for each directory, not per project.
> 
> For example you'll have this structure
> 
> ```
> /a/directory/path
> |____file1.fcstd
> |____file2.fcstd
> |____.fcbak
>          |____file1.fcstd1
>          |____file1.fcstd2
>          |____file2.fcstd1
> ```

I know this is an old response, but since this is still an open request... I personally would disagree with this solution because it doesn't really solve the "directory bloat" issue. It just hides it from view. The files are still there to be dealt with, taking up space, albeit a small amount, yes, but....

I would love to see this implemented as well, but with a central directory used for all of the autosaves and backup files.

adrianinsaval, 2023-06-12:
I agree the option of having all the backups in a central place should be available (that's why I advocated for allowing absolute paths in the config), but I think the default should be to place them in a subdirectory in the same path as the file.

Autosaved recovery files are unrelated and appropriately go to a system dependent temp directory. Backups are almost useless if they go to a temp directory and autosaves should definitely not be in a permanent storage directory.

MatthiasWM, 2023-06-28:
I am sorry, but I have to give this assignment back into the pool. Due to changes in the job, I won't have time to work on this. Apologies!

NXTler, 2023-08-26:
I would like to have this feature, just to have everything cleaned up

xgpt, 2024-02-01:
Agreed. The current situation is so messy! Please use a subdirectory! Hidden or not.

leoheck, 2024-05-30:
These backup files are annoying. I am constantly deleting them. I never used them. I have been working with Freecad for years already.

What is the status of this feature? Can we have this implemented?

Then the user could also have the settings to change the path of the bkp folder.
Which can be the current folder (argh!!) or another place since we should not care about the bkp files.

jflyer81, 2025-02-13:
Just wanted to draw attention to this issue again. Having the ability to specify a directory/path for the backup files would be extremely helpful. I don't need to see backup files basically ever (unless something's gone wrong), so having them clutter up my main models folder is a real pain. I'm leaning towards turning them off completely, but I hate not having any sort of backup for those occasions when things do get messed up.

oleglob71, 2026-09-14:
I strongly second the idea of storing backup files in a separate folder as they really make up a mess in a project directory, especially when the project contains lots of models spreaded over many different files. The issue could be solved either by creating a dedicated subdirectory in the project folder or, more preferable by mine, defining a special single folder for all backups, inluding autosavings. Both solutions would actually much help.

herrdeh, 2026-10-28:
I'd vote for leaving it to the user. So there could be 3 options in settings:
 
1. as it is today
2 save backups in a local hidden dir
3. save backups in a dir user can choose

I'd choose #3, I don't know if #2 really is desirable - who actually wants this, except @adrianinsaval  ?

FlachyJoe, 2026-10-28:
A hidden dir with its path relative to the project's one is great for a remote save / collaborative work.

maxwxyz, 2026-11-07:
This is available in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/1a9f440e6f92843da14e436374dd15f9fb06cdb3

You can choose an absolute backup directory or one relative to the document. Existing backups are left where they are.